Zolko-123/FreeCAD_Assembly4 #3: Can't run module

zackles, 2019-07-15:
Hello,
I tried to run your module with  :

- FreeCAD-asm3-Win64-Py3-Qt5-20190619
- FreeCAD_0.19.17337_x64_LP_12.1.2_PY3QT5-WinVS2015
- stable FreeCAD 0.18

Each time, I obtain the following error at the start of FreeCAD and the module is like missing : 

"During initialization the error "'charmap' codec can't decode byte 0x90 in position 1460: character maps to <undefined>" occurred in G:\Utilisateurs\me\Downloads\Freecad\Version\FreeCAD-asm3-Win64-Py3-Qt5-20190619\Mod\Mod_Asm4\InitGui.py
Please look into the log file for further information
"
Is there a specific version which Assembly4 work with?

Otherwise, your module seems great!
zackles

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.16830 +1504 (Git)
Build type: Release
Branch: LinkStage3
Hash: c80967b68820e135b93c6e69a5a538adf58da504
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/France (fr_FR)

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.17337 (Git)
Build type: Release
Branch: master
Hash: 0871fb49c71f1ac4029e441814332368f42bd78a
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/France (fr_FR)

OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16117 (Git)
Build type: Release
Branch: releases/FreeCAD-0-18
Hash: dbb4cc6415bac848a294f03b80f65e888d531742
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/France (fr_FR)

Zolko-123, 2019-07-23:
Thank-you for testing. This should be fixed now, can you please re-try ?

luzpaz, 2019-07-24:
@zackles please update and retest

zackles, 2019-07-26:
Sorry, github was configured with old mailbox. Just have seen your messages.

It is ok. It is fixed with all theses versions of freecad.

closed 2019-07-26 by zackles