Zolko-123/FreeCAD_Assembly4 #12: Please consider using Github Releases/Tags

luzpaz, 2019-11-25:
https://help.github.com/en/github/administering-a-repository/creating-releases

It's easier to follow development and can attract more users

Zolko-123, 2019-11-26:
Hi, I added a release, the snapshot of the current version. Is this what you had in mind ?

luzpaz, 2019-11-26:
Yes, this is great. BTW, I think MacOS uses `~/.FreeCAD/Mod/` as well (like Linux)

Zolko-123, 2019-12-02:
Hi,

may-be I misunderstand some things: what is the difference between Tags and Releases ? 

Specifically, I'd like to have a "Stable" "release" (if that's the correct name) which would always be referred-to by the Addon Manager, and have the "Development" main repository with the latest stuff that would be the rolling master branch (I might confuse names) 

Is it possible to have a "Stable" release, but that points to a Tag, and the value of this Tag can be updated ? Thus, I could set "Tags" when I think it's ready (v0.7.x...) and I would always point the "Stable" release towards the latest "Tag", and still work and update the master / development branch as I see fit.

This way, the Addon Manager could point to the "Stable" release of Assembly4, which would prevent people seeing every minor update I do, and only update when there really are some news.

Do I understand that correctly ?
Could you may-be give some advice on how to do that, and/or suggest a better organisation ?

Cheers,

Zoltan

luzpaz, 2019-12-02:
I'm not an expert on this and don't know enough on the subject to advise you what is the best approach. What I've seen is that people use a `master` branch as their main (default) branch and then have a development branch they develop on. When the time comes they merge the development branch in to master. Then continue hacking on the development branch until the time comes to merge again. 

Tags vs Releases are pretty much the same thing (there are [differences](https://git-scm.com/book/en/v2/Git-Basics-Tagging)) but in general Releases are glorified tags. They are a way to record milestones in the code that you can easily refer back to.

Zolko-123, 2019-12-03:
I created a branch called "development" and I'll make all updates there, and when time comes I'll merge them into "master". This means that "master" will be untouched and the merge should be easy. I have to update the examples and tutorials, so it's not really in the workbench code anyway.

Let's how this goes. Thank-you for your help.

luzpaz, 2019-12-03:
I'll keep an eye out to see if there is a better way to do this... or perhaps someone can weigh in. I didn't think about the issue with links breaking in documentation because of developing in a separate branch. I wonder how devs get around that?

closed 2019-12-05 by Zolko-123