Zolko-123/FreeCAD_Assembly4 #27: Initial feedback after going through Tutorial 1

justinclift, 2020-01-16:
Two main (potential) issues jump out at me after having gone through Tutorial 1:

* The initial picture showing the sketch to draw is missing the sizing for several dimensions:

[image]

The sketch I created had the stated dimensions the same, however something has gone wrong somewhere as my "finished" part looks like this instead of having the 2 bearings being correctly located:
[image]

Guessing the problem is my sketch was "too short" in some dimension.  If that's the case, then perhaps _Sketch_Axis.png_ should be updated to show the size for all dimensions?

* After adding a chamfer to the _axis_ Body in `asm_tuto1`, saving doesn't appear to work.  The error `<App> Document.cpp(2223): Partial loaded document 'axis' cannot be saved` is shown in the Report view panel.

Re-opening the documents... and yep, the chamfer wasn't saved. :frowning:

Zolko-123, 2020-01-17:
re-saving: did you have a window open on the part axis, along with the window open on the assembly ? I think that that doesn't work well with App::Link. If it's confirmed, I think it's a bug in App::Link.

Zolko-123, 2020-01-17:
> The sketch I created had the stated dimensions the same

are you sure that the dimension with "18mm" was relative to the vertical sketch axis ?

justinclift, 2020-01-17:
> re-saving: did you have a window open on the part axis, along with the window open on the assembly ? I think that that doesn't work well with App::Link. If it's confirmed, I think it's a bug in App::Link.

No. It's pretty simple to replicate.  Happens 100% reliably:

1. Start FreeCAD
[image]

2. Open `asm_tuto1.FCStd`
[image]

3. Make the `axis` body inside `asm_tuto1` active
[image]

4. Select the edge to chamfer
[image]

5. Add a 1mm chamfer to it (and press OK)
[image]

6. Click the Save document icon
[image]

7. Click `Yes` to also save the dependent files
[image]

** Note the warning that appears in the Report view**
```
<App> DocumentObject.cpp(699): Changes to partial loaded document will not be saved: axis#Chamfer001.Label
1.9e-07 <App> Document.cpp(2223): Partial loaded document 'axis' cannot be saved
```

8. Close the open documents (*File* -> *Close All*) then reopen `asm_tuto1.FCStd`.
[image]

The added chamfer isn't present. :frowning:

justinclift, 2020-01-17:
Just checked, this also happens with freshly compiled FreeCAD too (as of a few minutes ago):

```
OS: openSUSE Leap 15.1 (KDE//usr/share/xsessions/default)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.19299 (Git)
Build type: Unknown
Branch: test-v1
Hash: a285e89b2708de427fdce5b3b981a3ef630f5a4d
Python version: 3.6.9
Qt version: 5.9.7
Coin version: 3.1.3
OCC version: 7.3.0
Locale: English/UnitedKingdom (en_GB)
```

justinclift, 2020-01-17:
The "Branch: test-v1" is somewhat misleading.  It's actually `master`, I just created created an initial branch to play around in, not realising the branch name would show up here. :wink:

Zolko-123, 2020-01-17:
Strange, for me it works as advertised.

justinclift, 2020-01-17:
Hmmm.  Does it work for you with today's build of FreeCAD?

https://github.com/FreeCAD/FreeCAD/releases/tag/0.19_pre

Asking "just in case". :smile:

Zolko-123, 2020-01-17:
yes, also with that one. There is an option in the preferences about "disabling partially loaded documents". For me it's unchecked, may-be you can look at that setting and change-it, see if that has an influence. Also, in the PartDesign preferences, I have checked all 3 options "Automatically refine model after..."

OS: Ubuntu 18.04.3 LTS (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.19285 (Git) AppImage
Build type: Release
Branch: master
Hash: 8b2e318f15ee6304e8d89f48b2ac45af3bf61aed
Python version: 3.8.1
Qt version: 5.12.5
Coin version: 4.0.0a
OCC version: 7.4.0
Locale: English/United Kingdom (en_GB)

justinclift, 2020-01-17:
Interesting.  Tried those 3 options in the PartDesign Preferences but they made no difference.

Then tried turning on the `Disable partial loading of external linked objects` option in **General > Document**.
[image]

That worked.  eg enabling that option then means the referenced documents "fully load", so any  changes are kept when saved.

Sounds like a note about this should be added somewhere appropriate. :smile:

rschader, 2020-01-26:
I am wondering if this is the appropriate place to report some issues I am having with assembly4 on 0.19pre on Windows 10, trying to go through this same tutorial from scratch. I am fairly new to freecad (been waiting for a good assembly module).  Here is my PC info:

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.19358 (Git)
Build type: Release
Branch: master
Hash: 9e736095187450812e16ff9b92b6839e578f5506
Python version: 3.7.3
Qt version: 5.12.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/United States (en_US)

Assembly4 version Current version: 0.7.12, 2020-01-25, apparently just updated to this version, but the problem still persists!

Here is what is happening to me: When I try to edit datum on the bearing part, and hit OK after selecting the circular edge and specifying concentric, when I hot the OK button I get the following error:
Illegal storage access...
Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Illegal storage access! Please save your work under a new file name and restart the application!
If I hit the OK button again, Freecad crashes. If I hit cancel, it appears that the LCS was modified, but then Freecad crashes shortly afterwards. I've tried several different options in the bearing file, such as import as assembly when reading the step file, or not, I recently tried disabling the partial loading as suggested above, to no avail. It kind of sounds like a permission problem?

justinclift, 2020-01-26:
> I am wondering if this is the appropriate place to report some issues I am having with assembly4 ...

Kind of yes, kind of no. :wink:

In theory, "best practise" would be to create a new issue here in this repository, with the same info you gave.  But, some places aren't real fussed about it so you may be fine. :smile:

Zolko-123, 2020-01-27:
@rschader: yes, this is a recently discovered bug in FreeCAD. For the time being, don't use the "Edit datum" command to edit the attachment, but rather the edit MapMode in the properties of the LCS

rschader, 2020-01-28:
Thanks! That works. I just downloaded the latest 0.19 release and see that bug is still there!
OS: Windows 7 SP 1 (6.1)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.19373 (Git)
Build type: Release
Branch: master
Hash: c47512b7b2b3b64b4a21a86ad34a565459a13e6a
Python version: 3.7.3
Qt version: 5.12.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/United States (en_US)

closed 2020-02-18 by Zolko-123