Zolko-123/FreeCAD_Assembly4 #30: Feedback on Tutorial 2

arcanemachine, 2020-01-18:
Hi, just made note of a couple things while going through the animation tutorial (Tutorial 2):

1. When creating the LCS's for the master sketch, the tutorial says to "call it LCS_arm and attach it to the line in the master sketch representing ... the arm."

I was not able to get the model to align correctly unless I selected the farthest vertex from the origin (Sketch_master.Vertex3 in my document) and also the farthest line from the origin (Sketch_master.Edge2 in my document), then select attachment mode "Align O-X-Y" (just like the previous LCS).

It may seem obvious but I feel it's worth pointing out. My first time going through the tutorial, I was just robotically following the instructions and my "arm" object didn't line up correctly. It took me a second try before I understood the "bigger picture" of what was happening in each step.

2. A couple times throughout the tutorial, I found myself unsure if I had set my LCS's properly. Each time, however, I was able to compare mine to the ones in the pictures to ensure they were set correctly. Perhaps you could state explicitly at the beginning that readers can compare their LCS's against your pictures to ensure they are set up correctly, if they are unsure.

3. The variable "Dist_pivot" turns into "Disk_pivot" about halfway into the tutorial (in the "Base Part" subsection of the "Parts" section of the tutorial). A search of the document yields several instances of both terms... as well, the [third picture](https://github.com/Zolko-123/FreeCAD_Assembly4/blob/master/Examples/Tutorial2/tuto2_addVariable.png) in the tutorial has both terms. I assume this an error. (This may have led to minor confusion for me :D )

All in all, a decent tutorial (if not a little overwhelming). I feel that a substantial amount of material could be cut out to make things more concise for simpletons such as myself... if time permits I may throw together a quick tutorial of my own.

Thank you for this turorial. The feeling I got when watching the assembly animation for the first time was... ahem... NSFW.

Zolko-123, 2020-01-19:
I corrected Disk_pivot things, except for the screenshot, I'll see when I get to it

> The feeling I got when watching the assembly animation for the first time was... ahem... NSFW.

Could you elaborate ?

arcanemachine, 2020-01-19:
Sorry... I was joking that I became "sexually aroused" the first time I watched the animation come to life. It was my first time seeing a FreeCAD animation come to life, and the process was very simple, so I was excited to see it. I'll edit my comment.

Zolko-123, 2020-01-19:
ah, OK, thank-you. I close the issue then.

closed 2020-01-19 by Zolko-123