Zolko-123/FreeCAD_Assembly4 #41: Crash modifiying LCS

jlz3008, 2020-02-08:
Using Assembly 4. Last updated 07-02-2020

File->New
Assembly->New Model
Close -> Save -> Reopen
Select model
Assemby->New Body
Double click in Body
Create Schetch
Select XY plane
Draw a Rectangle -> Close
Pad ->Acept
Select model
New Coordinate System -> Select random Edge -> Acept
Select LCS_1
Edit datum -> Select Another Edge -> Accept ->Crash

OS: Ubuntu 18.04.3 LTS (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19. Last daily On 7-02-2020
Build type: Release
Python version: 3.6.9
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: Spanish/Spain (es_ES)

Zolko-123, 2020-02-08:
Yes, this is a known FreeCAD bug:

https://tracker.freecadweb.org/view.php?id=4258

closed 2020-02-08 by Zolko-123

jlz3008, 2020-02-10:
But it  doesn't happen if I repeat de test using Part, Body and Create LCS  from Part Design "standard".
Only It happen if I use Assemby4 tools.