Zolko-123/FreeCAD_Assembly4 #52: Add LGTM badges

luzpaz, 2020-03-16:
ref: https://github.com/FreeCAD/FreeCAD-addons/issues/163

Zolko-123, 2020-03-17:
thank-you for your contribution, but this was 1 step in the wrong direction: while it's nice to have some automatic code analysers, I refuse to acknowledge a "grade" from an automated system. We're not letting SkyNet control us.

closed 2020-03-17 by Zolko-123

luzpaz, 2020-03-18:
you don't have to use the grade badge then...just use the alert badge ?

Zolko-123, 2020-03-18:
On mercredi 18 mars 2020 02:56:40 CET luzpaz wrote:
> you don't have to use the grade badge then...just use the alert badge ?

hum, I dunno ... what is the benefit of this ? We want to make good FreeCAD code, not good Python code. 

cheers,

zoltan

-- 

═══════════════
   HUBERT Zoltán 
═══════════════

luzpaz, 2020-03-18:
@Zolko-123 feel free to discuss it on the [forum thread](https://www.forum.freecadweb.org/viewtopic.php?f=10&t=40228)

PR #53 (2020-03-17): README: Add LGTM badges
Closes #52  
ref: https://github.com/FreeCAD/FreeCAD-addons/issues/163
commit f0a62fcc (2020-03-17): README: Add LGTM badges

Closes #52  
ref: https://github.com/FreeCAD/FreeCAD-addons/issues/163