Zolko-123/FreeCAD_Assembly4 #62: Error by activating workbench assembly 4

AcumenDev, 2020-05-12:
By activating workbench assembly 4 I get an error in the console: 

`
No module named 'showCommand'
Traceback (most recent call last):
  File "<string>", line 97, in Initialize

`

Thanks for assembly 4.

FreeCAD version:
OS: Debian GNU/Linux 10 (buster) (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.20514 (Git) AppImage
Build type: Release
Branch: master
Hash: a05ffd499366ab12fa1b43f37db73f4f1e9c332d
Python version: 3.8.2
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

FreeCAD Assembly 4 workbench
Current version 0.9.2, 2020-05-08

AcumenDev, 2020-05-12:
Last Freecad, also  the error didn't go away
Word size of FreeCAD: 64-bit
Version: 0.19.21007 (Git) AppImage
Build type: Release
Branch: master
Hash: fdd99d2b2854d827320c7c952fddc3c12963d4eb
Python version: 3.8.2
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

Zolko-123, 2020-05-15:
yes, sorry, it was a test command that I forgot. I removed now and fixed

closed 2020-05-15 by Zolko-123