Zolko-123/FreeCAD_Assembly4 #63: [FR] Easy way of showing only selected subassembly/part

DerAndere1, 2020-05-13:
I would like to have a button/hotkey that allows to show only currently selected subassembly(ies). Or at least a hotkey to select the top-level App::Part container called "Model"

**Use case:** 
Focussing on certain parts of a big assembly ("Model") created with Assembly4 that contains multiple App::Links to subassemblies/parts (e.g. https://github.com/Zolko-123/FreeCAD_Examples/tree/master/Asm4_Example1). 
Existing macros for showing only selected objects (https://wiki.freecadweb.org/Macro_Visibility_Manager and https://wiki.freecadweb.org/Macro_Toggle_Visibility2_1-2) do not work well with App::Link. I have to select both, the top-level App:Part "Model" container of the main assembly  (from combo tree view) AND the desired subassembly (Select Link in combo treeview or select in 3D view). Searching for the top-level Model container is annoying and unintuitive.

Zolko-123, 2020-05-15:
I'm sorry but don't understand the feature you're asking for

DerAndere1, 2020-05-15:
For example in the Assembly4 assembly asm_tuto1.FCStd from https://github.com/Zolko-123/FreeCAD_Examples/tree/master/Asm4_Example1 I would be able to do the following:

i) select parts "axis" and "bearing3" (form the 3D-view or by clicking on the corresponding App::Links in the combo tree view using Ctrl + left mouse button).
ii) Click on a new button (that has to be implemented) which should make visible only the features that are selected from step i (in this case "axis" and "bearing3") while hiding all other features in the 3D view

Zolko-123, 2020-06-01:
ah OK, I understand now. Hum, let me think about that ...

Zolko-123, 2020-06-26:
Sorry, but I didn't find a solution for this problem. For now, I'll close this request.

closed 2020-06-26 by Zolko-123

DerAndere1, 2020-07-13:
Fixed if I use [realthunder's FreeCAD 0.19 release 0.11 2020.07.06](https://github.com/realthunder/FreeCAD_assembly3/releases/tag/0.11), after instaling/updating the assembly4 workbench using Tools -> Addon manager -> Assembly4. So the fix seems to exist in one of the changes in realthunder's LinkStage3 branch. I guess the same underlying problem causes issue https://github.com/Zolko-123/FreeCAD_Assembly4/issues/79.