Zolko-123/FreeCAD_Assembly4 #64: [FR] preselecting LCS in 3D view for changing attachment

DerAndere1, 2020-05-13:
I would like to be able to change attachment of subassemblies/parts to the parent Model using the following workflow: 
1. Make visible all LCS of the assembly
2. Select one LCS of the subassembly and one LCS of the parent Model
by holding down Ctrl while clicking on both LCS **in the 3D view**
3. Click on the button "Move/Attach Part in the assembly". This should open the dialog "Place linked Part", but with those two LCS **preselected**.

**Usecase:**

Choosing attachment LCS visually in 3d view. Especially helpful for assembling parts with many LCS where choosing from a list of LCS becomes guesswork

DerAndere1, 2020-05-13:
I have no FreeCAD sripting experience but here is some rough pseudocode proposal for the "Place linked Part" dialog: 
```py
selection = FreeCADGui.Selection.getSelectionEx() # see https://wiki.freecadweb.org/Selection_API
sel_subobj_part0=selection[0].SubObjects # TODO: Find documentation about getSelectionEx()[0].SubObjects
sel_subobj_part1=selection[1].SubObjects

selected_LCS_part0 = [sel_SubObj for sel_SubObj in sel_subobj_part0 if (type(sel_SubObj) is PartDesign.CoordinateSystem)] # TODO: Is the type of an LCS really PartDesign::CoordinateSystem and what is the correct Python syntax?
selected_LCS_part1 = [sel_SubObj for sel_SubObj in sel_subobj_part1 if (type(sel_SubObj) is PartDesign.CoordinateSystem)] 

# Existing code for the "Place linked Part" dialog continues here ... . At the end of that code insert something like this:

place_dialog.LCS_part0_ListWidget.setCurrentIndex(selected_LCS_part0[0].Name) # TODO: Replace this with actual code that preselects the correct element from the left list widget in the "Place linked Part" dialog.
place_dialog.LCS_parentAssembly_ListWidget.setCurrentIndex(selected_LCS_part1[0].Name) # TODO: Replace this with actual code that preselects the correct element from the right list widget in the "Place linked Part" dialog.
```

Zolko-123, 2020-05-15:
yes, this would be a good feature, but I don't know how to implement it.

closed 2020-05-15 by Zolko-123

DerAndere1, 2020-05-15:
If you keep this feature request open and say you like it, someone else might implement it and create a pull request. That is how these cases are handled in most open source projects on github. You may use labels to mark this as an enhancement request so that you can filter it out from your issue list. Thanks so much for Assembly4!

DerAndere1, 2020-11-06:
pull request 137 implements this feature as good as possible
The GUI of FreeCAD currently provides no way to multi-select two LCS from the 3D-View. Thus the workflow for this feature will be as follows:

1. In the 3D-view, double-click an LCS of the linked part
2. In the 3D-view, (Ctrl + left-click) on an LCS in the parent_part that you want to use as the attachment point
3. Hold Ctrl while double-clicking in the **combo-view Model tree** on the entry that was highlighted due to step 2.
4. Left-click button "Edit Placement of Part" (Tooltip: "Move/Attach a part in the assembly") to open the dialog "Place Linked Part"
5. Observe that the two LCS that were selected in steps 1 - 3 are also preselected in the dialog