Zolko-123/FreeCAD_Assembly4 #72: Propagate LCS from sub assembly to assembly

zaphodbe, 2020-06-01:
I have created an assembly to model the platter of a typewriter.  It consists of an axel with the plastic platter as two sub-assemblies each with a single body/part.  Each of those sub-assemblies have several LCS's defined to become attachment points.  How do I propagate those sub=assembly LCS's to the parent assembly.  

Also if I try just creating new LCS's for the full assembly it won't allow me to use the features from the sub-assembly to select where to put the new LCS's

TIA

Zolko-123, 2020-06-01:
Did you try the "Import Datum" command ? It was meant to deal with that scenario.

- in the assembly document, select, in the model tree, the datum in the child-part/sub-assembly that you want to propagate
- use the "Import Datum" command
- this should have created a datum of the same type with the same name, in the assembly.

The imported datum needs to be at the root of the child part/sub-assembly, it can't work on datums that are nested inside a part inside the sub-assembly. For that, open the sub-assembly, and import the datum from the child part into the sub-assembly, and **then** into the assembly.

For now, this is a little tedious, I know.

zaphodbe, 2020-06-02:
That worked thanks.

closed 2020-06-02 by zaphodbe