Zolko-123/FreeCAD_Assembly4 #73: FreeCAD crashes on import

dstuessy, 2020-06-13:
Importing a part in Assembly 4 seems to cause my FreeCAD instance to crash.

Any ideas how I can get around this?

I get the following output in the terminal:

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3ef20) [0x7f7ebb63cf20]
#1  0x7f7ec112c131 in App::OriginGroupExtension::relinkToOrigin(App::DocumentObject*) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libFreeCADApp.so+0x11
#2  0x7f7ec112d107 in App::OriginGroupExtension::addObjects(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> >) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libFreeCADApp.so+0x37
#3  0x7f7ec10dc717 in App::GroupExtension::addObject(App::DocumentObject*) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libFreeCADApp.so+0x97
#4  0x7f7ec10dc3d9 in App::GroupExtension::addObject(char const*, char const*) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libFreeCADApp.so+0x59
#5  0x7f7ec10df98e in App::GroupExtensionPy::newObject(_object*) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libFreeCADApp.so+0x4e
#6  0x7f7ec10df9f7 in App::GroupExtensionPy::staticCallback_newObject(_object*, _object*) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libFreeCADApp.so+0x27
#7  /tmp/.mount_freecasfpOfl/usr/bin/../lib/libpython3.6m.so.1.0(_PyCFunction_FastCallDict+0x18e) [0x7f7ec0725c1e]
#8  /tmp/.mount_freecasfpOfl/usr/bin/../lib/libpython3.6m.so.1.0(+0x1650da) [0x7f7ec07bf0da]
#9  /tmp/.mount_freecasfpOfl/usr/bin/../lib/libpython3.6m.so.1.0(_PyEval_EvalFrameDefault+0x3002) [0x7f7ec07c2902]
#10  /tmp/.mount_freecasfpOfl/usr/bin/../lib/libpython3.6m.so.1.0(+0x164370) [0x7f7ec07be370]
#11  /tmp/.mount_freecasfpOfl/usr/bin/../lib/libpython3.6m.so.1.0(_PyFunction_FastCallDict+0x2c6) [0x7f7ec07c8196]
#12  /tmp/.mount_freecasfpOfl/usr/bin/../lib/libpython3.6m.so.1.0(_PyObject_FastCallDict+0x17e) [0x7f7ec06ccafe]
#13  /tmp/.mount_freecasfpOfl/usr/bin/../lib/libpython3.6m.so.1.0(_PyObject_Call_Prepend+0xce) [0x7f7ec06ccbee]
#14  /tmp/.mount_freecasfpOfl/usr/bin/../lib/libpython3.6m.so.1.0(PyObject_Call+0x6a) [0x7f7ec06cc8aa]
#15  0x7f7ec1a4725e in Gui::TaskView::TaskDialogPython::accept() from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libFreeCADGui.so+0x14e
#16  0x7f7ec1a3f643 in Gui::TaskView::TaskView::accept() from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libFreeCADGui.so+0x43
#17  0x7f7ebc8178d1 in QMetaObject::activate(QObject*, int, int, void**) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libQt5Core.so.5+0x2c1
#18  /tmp/.mount_freecasfpOfl/usr/bin/../lib/libQt5Widgets.so.5(+0x285d10) [0x7f7ebd2c3d10]
#19  0x7f7ebc8178d1 in QMetaObject::activate(QObject*, int, int, void**) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libQt5Core.so.5+0x2c1
#20  0x7f7ebd296282 in QAbstractButton::clicked(bool) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libQt5Widgets.so.5+0x32
#21  /tmp/.mount_freecasfpOfl/usr/bin/../lib/libQt5Widgets.so.5(+0x25847a) [0x7f7ebd29647a]
#22  /tmp/.mount_freecasfpOfl/usr/bin/../lib/libQt5Widgets.so.5(+0x2594ec) [0x7f7ebd2974ec]
#23  0x7f7ebd297664 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libQt5Widgets.so.5+0xd4
#24  0x7f7ebd1de73a in QWidget::event(QEvent*) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libQt5Widgets.so.5+0x96a
#25  0x7f7ebd19bc6c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libQt5Widgets.so.5+0x9c
#26  0x7f7ebd1a0b5d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libQt5Widgets.so.5+0xc6d
#27  0x7f7ec1876371 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libFreeCADGui.so+0x41
#28  0x7f7ebc7ed425 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libQt5Core.so.5+0x75
#29  0x7f7ebd19f860 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libQt5Widgets.so.5+0x1a0
#30  /tmp/.mount_freecasfpOfl/usr/bin/../lib/libQt5Widgets.so.5(+0x1b9cd1) [0x7f7ebd1f7cd1]
#31  /tmp/.mount_freecasfpOfl/usr/bin/../lib/libQt5Widgets.so.5(+0x1bc2c3) [0x7f7ebd1fa2c3]
#32  0x7f7ebd19bc6c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libQt5Widgets.so.5+0x9c
#33  0x7f7ebd1a02ea in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libQt5Widgets.so.5+0x3fa
#34  0x7f7ec1876371 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libFreeCADGui.so+0x41
#35  0x7f7ebc7ed425 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libQt5Core.so.5+0x75
#36  0x7f7ebcb917db in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libQt5Gui.so.5+0x40b
#37  0x7f7ebcb92f75 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libQt5Gui.so.5+0x115
#38  0x7f7ebcb7474b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libQt5Gui.so.5+0x7b
#39  /tmp/.mount_freecasfpOfl/usr/lib/libQt5XcbQpa.so.5(+0xa05d0) [0x7f7eb45845d0]
#40  /tmp/.mount_freecasfpOfl/usr/lib/libglib-2.0.so.0(+0x569be) [0x7f7eb83419be]
#41  /tmp/.mount_freecasfpOfl/usr/lib/libglib-2.0.so.0(g_main_context_dispatch+0x33) [0x7f7eb8342826]
#42  /tmp/.mount_freecasfpOfl/usr/lib/libglib-2.0.so.0(+0x57a0b) [0x7f7eb8342a0b]
#43  /tmp/.mount_freecasfpOfl/usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x4a) [0x7f7eb8342acf]
#44  0x7f7ebc83e98c in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libQt5Core.so.5+0x5c
#45  0x7f7ebc7eb53b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libQt5Core.so.5+0xfb
#46  0x7f7ebc7f3466 in QCoreApplication::exec() from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libQt5Core.so.5+0x86
#47  0x7f7ec181325c in Gui::Application::runApplication() from /tmp/.mount_freecasfpOfl/usr/bin/../lib/libFreeCADGui.so+0x16bc
#48  /tmp/.mount_freecasfpOfl/usr/bin/FreeCAD(main+0x6c9) [0x403829]
#49  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7) [0x7f7ebb61fb97]
#50  /tmp/.mount_freecasfpOfl/usr/bin/FreeCAD() [0x404699]
```

Zolko-123, 2020-06-14:
What are your FreeCAD and Assembly4 versions ?

Zolko-123, 2020-06-26:
No reply ? Closed

closed 2020-06-26 by Zolko-123