Zolko-123/FreeCAD_Assembly4 #78: [enhancement] dialog for setting attachment

donovaly, 2020-06-25:
After placing a part one often needs to change the attachment relative to the LCS. This has to be done in the Data tab, section Assembly -> Attachment.

However, it would be nice if there would be a dialog to do this. I think A4 could use the existing Edit Datum dialog for this task.

Zolko-123, 2020-06-26:
you mean: after attaching a part to an LCS in the assembly, you would want to change the AttachmentOffset directly in the placement dialog ?

donovaly, 2020-06-26:
I would like to right-click on the part in the tree and in the context menu there should be under "Assembly" an entry "Edit Attachment". This should open the attachment dialog. I mean the same dialog that I already get when double-click on an LCS in the tree:
[image]

There I can easily and especially immediately visibly change the attachment. The reason is that it is often quicker not to add another LCS but to use an existing LCS and attach to this with an offset.

Zolko-123, 2020-06-26:
> right-click on the part in the tree

but Parts don't have attachment, only datum objects have them ! You cannot open that UI for parts and bodies. Actually, datums, sketches, Part features have attachment (MapMode)... but not Parts, Bodies, and Links: they don't have Attachments, only Placements.

donovaly, 2020-06-26:
> You cannot open that UI for parts and bodies.

Not yet. Therefore I labeled this issue as enhancement. I want to have this feature added to assembly4. It is not necessary to make this a feature for other WBs.

Zolko-123, 2020-06-26:
Unfortunately, this is a core FreeCAD functionality, nothing that Asm4 can do about. I close this issue

donovaly, 2020-06-26:
> Unfortunately, this is a core FreeCAD functionality, nothing that Asm4 can do about. I close this issue

I don't understand. The attachment dialog exists in FC. Why can't one call it whenever one likes?
The dialog is the file
https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Part/AttachmentEditor/TaskAttachmentEditor.ui
and it can be called via Python (I assume) using this file:
https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Part/AttachmentEditor/TaskAttachmentEditor.py

So since it is part of the Part WB there should be a way to use it. Maybe I have a thinko. Please tell me.

Zolko-123, 2020-06-27:
On vendredi 26 juin 2020 21:26:07 CEST Uwe wrote:
> > Unfortunately, this is a core FreeCAD functionality, nothing that Asm4 can do about. I close this issue
> 
> I don't understand. The attachment dialog exists in FC. Why can't one call it whenever one likes?

you can, with the command :

# Gui.runCommand('Part_EditAttachment')

try it for yourself.

zoltan

> The dialog is the file
> https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Part/AttachmentEditor/TaskAttachmentEditor.ui
> and it can be called via Python (I assume) using this file: [...]

-- 

═══════════════
   HUBERT Zoltán 
═══════════════

donovaly, 2020-06-28:
> Gui.runCommand('Part_EditAttachment')

Yes. But you still misunderstand me. It is not that you should call it as it is. There is no attachment, but only an alignment to be done.
So I thought about this:
- take the relevant UI elements from the existing
https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Part/AttachmentEditor/TaskAttachmentEditor.ui
- put it to a new .UI file that you can call via Assembly 4.

So all I want is a dialog instead the need to edit the properties.

I am not a Python Gui, but I once fixed some issues in the dialogs of A2plus.  Thus I learned that Python can access .UI files. (but not much more about how exactly.)

donovaly, 2020-06-28:
Can you please reopen this. In case you don't see the need of having a dialog, I will have a look on a rainy day if I could do this. Thus I need this open as reminder.

Zolko-123, 2020-06-29:
OK, I don't completely understand, but let's see what it brings

abetis, 2020-08-23:
I agree that for better usability the attachment settings should have better access than it is today.
That's the first thing I go to change after linking an object into the assembly.
I have to click 2-3 times in the data tab, since the assembly placement tree constantly collapse when clicking working, clicking in the view and so on...
If there will be immediate update during the change of parameters, that will easy even more on the use-case.

Zolko-123, 2020-08-23:
thinking about this, it would be quite easy to add the 3 translation offsets in the place part dialog. 

But adding rotational offsets will not be possible, because the rotations are already used by the Rot-X ... buttons, therefore it's not possible (or very tricky) to add a random rotation there. If this is necessary, an intermediate LCS should be used.

abetis, 2020-08-24:
I think setting the offsets in the will cover 80-90% use cases.
Cant you add an edit box close to the rotation buttons that will specify the amount to rotate?
By default it could be 90deg, but users would have the possibility to change it.

Zolko-123, 2020-11-15:
I think this can be closed now

closed 2020-11-15 by Zolko-123

donovaly, 2020-11-22:
Many thanks @DerAndere1 ! This improves the workflow a lot.