Zolko-123/FreeCAD_Assembly4 #79: LCS of models cannot be made invisible

donovaly, 2020-06-25:
I have an assembly containing other files as models and as parts.

Since I need the final assembly to be nicely rendered I tried to hide all LCS. This works for the parts, but for the models there is always one LCS that cannot be hidden.

The bug is that the command to hide the LCS goes to the global LCS, see this screencast.
[image]

So you see, pressing spacebar at the desired model LCS hides/shows instead the global LCS.

I use A4 0.9.6 and
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21769 (Git)
Build type: Release
Branch: master
Hash: 801bfe4ba5f0910e9b2885917c521ab7fd8e9020
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)

donovaly, 2020-06-26:
As workaround I can open the file containing the model, in my case the file "CPP1_Bodenplatte". There I hide the LCS and then also in the assembly the LCS is hidden.

So I think the bug is that Assembly 4 does not update the external dependencies fully for the visual state of LCS for parts included as model.

Zolko-123, 2020-06-26:
I think this is actually an App::Link bug/feature, because Asm4 only uses the raw interface, there's nothing custom about it. If you could make a minimal model that shows the behaviour, then we could ask realthunder what he thinks about it.

donovaly, 2020-06-26:
> If you could make a minimal model that shows the behaviour
Here it is:
[A4-example-LCS-hiding.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/4837242/A4-example-LCS-hiding.zip)

Open the assembly file, there go to the "CPP1-Bodenplatte", select the different LCS in the tree and press spacebar.
Result: the main LCS "LCS_Bodenplatte" cannot be hidden because the command is applied to the main LCS of the overall model "LCS_Origin".

Zolko-123, 2020-06-26:
thank-you, I'll look at it

DerAndere1, 2020-07-11:
I was not able to reproduce this with [realthunder's FreeCAD 0.19 release 0.11 2020.07.06](https://github.com/realthunder/FreeCAD_assembly3/releases/tag/0.11), after instaling/updating the assembly4 workbench using Tools -> Addon manager -> Assembly4

abetis, 2020-09-07:
I can confirm that the problem exist in my last build:
```
OS: Ubuntu 18.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22292 (Git)
Build type: Unknown
Branch: master
Hash: f09d48d0fe00590997b88b52b610210ac606951c
Python version: 3.6.9
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```
I work mostly with imported STEP files, so it's already a habit for me to define the LCSs and hide them in the model before I save it.

donovaly, 2020-09-15:
> Version: 0.19.22292 (Git)

I have now
Version: 0.19.22452 (Git)

and with this, this issue seems to be fixed.
Can you please check if this is fixed for you as well with today's Git master?

DerAndere1, 2020-10-01:
this issue is fixed for me with FreeCAD_0.19.22522_Win-Conda_vc14.x_x86_64 from https://github.com/FreeCAD/FreeCAD/releases under Windows 10 64 bit.

donovaly, 2020-10-02:
> this issue is fixed for me

Good. Marking this fixed.

closed 2020-10-02 by donovaly