Zolko-123/FreeCAD_Assembly4 #80: Edit placement idalog is too wide

donovaly, 2020-06-26:
After inserting a part, I use the menu  Assembly->Edit Placement of Part. The appearing Task dialog is much too wide:

[image]

This suddenly happened in my assembly after I already called this dialog for other parts and that time the dialog had the correct size.
I restarted FreeCAD but the problem remained.

I found the problem:
There is one part in the assembly with a large name:
[image]

So it is correct that the Task dialog is that wide. But for the other parts, the name is much shorter thus the dialog should fit to the length if the name.
This is currently not the case because I can for example made the dialog so small that one cannot read the full name:
[image]

I use Assembly 0.9.6 and
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21769 (Git)
Build type: Release
Branch: master
Hash: 801bfe4ba5f0910e9b2885917c521ab7fd8e9020
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)

Zolko-123, 2020-06-26:
ah, OK, yes: I think it's because of a default Qt behaviour that stretches a widget to the text length. I'll set a maximum length (but then you'll have to scroll to see the name)

donovaly, 2020-06-26:
> I'll set a maximum length

This would not be right. Instead the dialog should be as wide as necessary to contain the content. I only know the C++ connection to Qt and there you set this via the UI file.

I opt for this because it is quicker to move the splitter to the side to see the dialog in full size than to scroll in its different textEdits.

donovaly, 2020-09-15:
> Any news on this ? Is this still a problem ?

Still a problem with 0.9.8 and latest FC master.

By the way, the release 0.9.8 does not appear here: https://github.com/Zolko-123/FreeCAD_Assembly4/releases

Zolko-123, 2020-09-16:
yeah, I didn't touch this lately.

kravietz81, 2020-11-20:
Can you strict widness of that dialog to some value and if name is too long show it as pop-up comment??

Did you consider wrapping of too long names??

Zolko-123, 2020-11-20:
fixed, please update and check

closed 2020-11-20 by Zolko-123

donovaly, 2020-11-22:
Many thanks. The bug is now fixed.