Zolko-123/FreeCAD_Assembly4 #81: Wrong warning in report panel on placing parts

donovaly, 2020-06-26:
I added a new part to the main model. In the placement dialog I am in this state:

[image]

Now I need to select the LCS and I need to start with one of the 2. But no matter if I select at first the linked LCS or the parent LCS, I always get immediately this warning: "Problem in selections" in the report panel.

This warning should only be issued if the selections are incompatible etc. As it is you get it in the normal use case.

Zolko-123, 2020-06-26:
Yes, I have seen that. will try to correct

Zolko-123, 2020-07-17:
I believe it's corrected now

closed 2020-07-17 by Zolko-123

donovaly, 2020-07-22:
I tested today and I still get the incorrect warning.