Zolko-123/FreeCAD_Assembly4 #82: [enhancement] Feature to replace parts

donovaly, 2020-06-28:
I started to try Assembly 4 for a real-life document to learn it a bit better.
In real-life I have the case the there are different versions of every part of an assembly. For example I have 3 different files for a part called "body". The reason is that at first we go parallel with different versions of an assembly. So "body-V1", "body-V2" etc. all fit perfectly into the assembly and only differs from a feature (e.g. a wider pocket).

So a really cool feature would be an option to replace a part in the assembly via a menu item. Of course it is the duty of the user to assure that the parts to be replaced must have the same LCS and LCS names.

Maybe you think this is overkill, and one should better create an assembly file for every version. I make this report because in reality it is not only one part with different version. In my case I have 3 of 7 parts in the assembly with each 3 versions. So 27 possible assemblies. During meetings it is helpful to play around with different combinations to reject some of the proposals. With the requested feature one would be able to do this more or less live during a meeting/discussion.

Zolko-123, 2020-06-29:
yes, good points.

But for presentations with several versions, what I do is to include them all and only show the one I want to demonstrate. This way it is also easy to compare them side-by-side, or by transparency.

OxygenLiu, 2021-01-02:
Thanks for the "configurations" feature! It works amazingly well!

donovaly, 2021-01-06:
> Thanks for the "configurations" feature! It works amazingly well!

Sounds great but where can I read to learn more about this feature?

OxygenLiu, 2021-01-11:
@donovaly in Assembly4 workbench, right click on "model", select "Assembly -> Save configuration",  then there will be a "Configurations" group under "Model" tree. 

You may manipulate the placement of some parts, and save another configuration, or restore a saved configuration. 

Good luck!

donovaly, 2021-01-11:
> @donovaly in Assembly4 workbench, right click on "model", select "Assembly -> Save configuration", then there will be a "Configurations" group under "Model" tree.

many thanks! This is a great feature.
All the great features need to be documented better. I will put this on my Todo list. (I once documented A2plus: https://wiki.freecadweb.org/A2plus_Workbench and would do so for Assembly4)

Concerning the initial enhancement request, I think it is not necessary to go so far to replace parts. For example, I linked 3 different versions of a part each as its own part. During the presentation, I simple show/hide the parts.

So if you agree, I would close this issue.

OxygenLiu, 2021-01-12:
Sounds good! We definitely need a good documentation of Assembly4.

donovaly, 2021-01-12:
> So if you agree, I would close this issue.

Doing so.

closed 2021-01-12 by donovaly