Zolko-123/FreeCAD_Assembly4 #84: Part LCS not listed in edit placement of part

panzergame, 2020-06-29:
I have created a part following the first tutorial, this part has a LCS in it :

[image]

Then i want to attach the part to LCS_Origin, but none part LCS are listed :

[image]

I don't know if i missed something or this is a bug.

The whole hierarchy looks like that :
[image]

Zolko-123, 2020-06-29:
the LCS_1 in charriotroulement is not in the App::Part but in the Body that is in the Part, Asm4 doesn't "see" that. You can do 2 things:
- move the LCS_1 to the root of the Part
- not use a Part at all, and use directly a Body, not wrapped in a Part (recommended).

If you're using PartDesign and Body, I recommend not to use a Part container. Use a Part container if you have multiple bodies/solids, or if you use the Part WB, or if you import STEP solids.

panzergame, 2020-06-29:
Indeed it works, thanks for the advice on part/body. 

If in understand correctly the LCS must be in the root object anyway that's why it can be in body or part, or it is an other reason ?

closed 2020-06-29 by panzergame

Zolko-123, 2020-06-29:
On lundi 29 juin 2020 23:35:18 CEST panzergame wrote:
> Indeed it works, thanks for the advice on part/body. 
> 
> If in understand correctly the LCS must be in the root object anyway that's why it can be in body or part, or it is an other reason ?

that's correct

zoltan
> 
> 

-- 

═══════════════
   HUBERT Zoltán 
═══════════════