Zolko-123/FreeCAD_Assembly4 #86: impossible to add LCS to parts in assembly

donovaly, 2020-06-29:
I have a part in my assembly and want to add a new LCS to it.
I cannot find a way to do this other than to open the file separately.

I think this is a bug because I can directly edit the part via PartDesign directly in the assembly. Theerfore is should also be possible to add an LCS directly in the assembly.

Zolko-123, 2020-06-29:
you mean: you have a part (Body ?) in an external file, you have inserted (via link) the part into the assembly, and you want to add an LCS to the external part through the link in the assembly ?

Yep, that might not work, because I didn't think of that. It's not a bug, it's a feature. 

Now, how important is that use-case ? It's already possible to modify a part from within the assembly (by modifying its parameters), but should a part be actually designed from within the assembly ? This might lead to unintended consequences.

donovaly, 2020-06-29:
> you mean: you have a part (Body ?) in an external file, you have inserted (via link) the part into the assembly, and you want to add an LCS to the external part through the link in the assembly ?

I have an assembly file containing a dozen parts. Each was once added via the "Insert part"" command. Now I realized, that to be able to add and place more parts, I need a bunch of new LCS. So I select the body in the tree but the "New Coordinate System" command is disabled. So I have to open the file of every body explicitly and add the LCS this way. This consumes much more time than to work directly in the assembly file. 

> it's a feature.

OK.

> Now, how important is that use-case ? It's already possible to modify a part from within the assembly (by modifying its parameters), but should a part be actually designed from within the assembly ?

That depends on the use case. I don't think that adding a LCS is a dramatic change since the body itself is not touched.
I can only speak for myself after my first meeting with/about it - it would have been helpful during the meeting.

Please note: the issues I report here are just reports that they won't be forgotten. Don't feel under pressure to act right now. I'm a developer too and sometimes I find time, sometimes not even for weeks.