Zolko-123/FreeCAD_Assembly4 #90: underscores when linking to part?

thoraxe, 2020-07-15:
This is most likely user error. When I try to insert a link to a part, the parts show up like so:

`_230303_assy#9230303 (Model)`

The filename is `9230303-assy` and it contains a single model called `9230303`.

I'm not sure why trying to link to the part ends up with the first character chopped and the dash converted to an underscore.

Zolko-123, 2020-07-17:
FreeCAD doesn't allow names to begin with a number.

closed 2020-07-17 by Zolko-123

thoraxe, 2020-07-17:
@Zolko-123 I think you are referring to:
https://wiki.freecadweb.org/Object_name

> The Name is assigned at the creation time of the object; afterwards it is no longer editable. The object can never be renamed.

I agree -- names cannot begin with a number. However, the user cannot create, modify, or change names. The user can only manage "labels", per this document.

In FreeCAD you cannot see the name for an object. You can only see its label. "Renaming" a part just re-labels it.

Am I misinterpreting the documentation?

Zolko-123, 2020-07-17:
On vendredi 17 juillet 2020 13:43:29 CEST Erik Jacobs wrote:
> @Zolko-123 I think you are referring to:
> https://wiki.freecadweb.org/Object_name
>
>
> I agree -- names cannot begin with a number. However, the user cannot create, modify, or change names. The user can [...]

I think you're correct. Except that at creation time, the Label is set to be the Name, therefore if you don't modify an object's label you do see it's name. Also, when passing the pointer over an object, you can see it's name in the status bar. The Name, not the Label.

cheers

zoltan

> 
> 

-- 

═══════════════
   HUBERT Zoltán 
═══════════════

thoraxe, 2020-07-17:
I think I figured it out. The issue is the _filename_.

While FreeCAD will happily let you save a file with a number as the first character, this is what ends up with an underscore. The objects inside then have internal names (eg: SketchXX) and Labels (eg: Profile).

Since my *filenames* are the same as the models in my parts, that's what makes it look weird.

So, Assembly4 is actually seeing the assembly *filename* "name" which is converted from a numerical first character to an underscore, as well as having hyphens converted to underscores.

So, as an example:

I have a file called 9230303-assy.FCStd
Inside it is a `Model` with the `Label` *9230303*.
Inside that is a `Body` with the original name of just `Body`.

When I try to insert a link to a part, Assembly4 is showing me the currently open *filenames*, but in its weirdo converted name format (where the first numerical character is replaced with an underscore and hyphens are replaced with underscores).

This gets us into a conversation about BOMs which I'll have to have with you, but I'll do it on the forum.

Thanks for sticking with me here!