Zolko-123/FreeCAD_Assembly4 #92: unknown command warnings

donovaly, 2020-07-27:
- start FreeCAD
- don't load any document, just change to the Assembly4 workbench

result: I get these warnings:
```
02:16:55  Unknown command 'Asm4_insertRod'
02:16:55  Unknown command 'Asm4_FSparameters'
```

mirage335, 2020-07-30:
Similarly, I see these warnings.

# Unknown command 'Part_Import'
# Unknown command 'Asm4_insertRod'
# Unknown command 'Asm4_FSparameters'

megadip, 2020-08-06:
I also get these warnings, but mine also says Unknown command 'Part_Import'.
I am uncertain what to do, I am quite new to linux in the scheme of things.

leTel, 2020-08-14:
Same issue
1. Start freecad-daily
2. Switch to Assembly4 workbench

```
11:51:48  Unknown command 'Part_Import'
11:51:48  Unknown command 'Asm4_insertRod'
11:51:48  Unknown command 'Asm4_FSparameters'
```

mirage335, 2020-08-30:
Maybe this should be pushed upstream to the AppImage developers. I was using the AppImage on a Debian system.

bkw777, 2020-08-31:
I get the same error as the OP. Not using AppImage in my case. But if the same thing happens in the AppImage, then you can't very well call it an "ubuntu screw-up".

donovaly, 2020-09-10:
I am only using Windows., so Linux distros are not to blame here ;-)