Zolko-123/FreeCAD_Assembly4 #93: Insert link to a part from a file starting with a number, will replace the starting number with "_"

abetis, 2020-08-15:
Hi,

If I have a file with a model, which starts with a number, say "6mm shaft", and I want to link it to the assembly, the name of the part shown in the "Insert a Part" dialog will appear as "_mm_shaft#Model" instead of expected "6mm_shaft#Model".
The name of the link (the field at the bottom of the same dialog) derives from that as well and will propose "_mm_shaft".

Having a number in the name in any other than the first letter, works ok.

Zolko-123, 2020-08-16:
yes, this is an internal FreeCAD feature.

closed 2020-08-16 by Zolko-123

abetis, 2020-08-16:
I would call it a bug and not a feature.
Maybe asm4 takes the parts name from a wrong field or something?

Zolko-123, 2020-08-16:
This is internal FreeCAD behaviour, nothing Asm4 can do about.