Zolko-123/FreeCAD_Assembly4 #95: Problem with linked parts when file is stored on a mounted volume

abetis, 2020-08-20:
Well, that might not be a common use case, but my working directory is stored on a mounted volume (on linux).
For example, if my home directory is in "/home/alex", the working directory "/home/alex/FreeCAD/test" is actually located on "/mnt/storage/FreeCAD/test"

I don't understand how the file location is handled by freecad and ASM4, but it looks like the path is not saved as "/home/alex/FreeCAD/test/filename.FCStd" as the user sees is when saving the file, but something like "../../../../../mnt/storage/FreeCAD/test/filename.FCStd"
At least that's what I can see from a "Link not restored" error in report view.

To make it even worse, the real physical path can be different on another PC where I open the files, but the logical name stays the same "/home/alex/FreeCAD/test".

Is there anything I can do to eliminate such problem?
For all the files I create them, save, close and reopen from the logical path before I link them to other files.

Zolko-123, 2020-08-20:
File path of linked files are stored as relative paths, because the idea is that a project will me self-contained, so that you can pass the entire directory to a collegue and he can use it. In this case, all parts in the assembly are supposed to be in subdirectories of the assembly. 

This covers many use-cases, but not all. 

For your particular use-case, do you think you could make a link to your mounted drive in the same directory as the main assembly is, so that FreeCAD will follow relative path ? And you arrange so that that link points to the correct location.

Or did I misunderstand something ?

abetis, 2020-08-21:
All the files are in the same directory.I start thinking that the problem started when I saved the same files on the second PC.On both PCs the files are in the same logical location "/home/alex/FreeCAD/test".
The FreeCAD directory is a symbolic link to another location, storage mount.
The difference of the physical location is in the mounting point, on one PC its:
"/media/storage/FreeCAD/test"
on another PC its:
"/mnt/storage/FreeCAD/test"

From what you're saying, the links should be ignorant from that difference since all the files are in the same direction, so all the paths should be "./file1", "./file2".
But as I see it from the error printout it looks like the path goes down to the mounting point and back up to the physical location:
"../../../../../mnt/storage/FreeCAD/test/filename.FCStd"

Is there a way to look into the files for the paths without opening them in FreeCAD?
I see that FCStd are binary files...

Zolko-123, 2020-08-21:
They are compressed text files, you can uncompress the archive with ark (in Linux/KDE).

I'll look into this, looks like a severe bug, thank-you for your report

abetis, 2020-08-25:
That issue blocks my work, even on my home PC where no NFS mounts exist, only symbolic link to locally installed RAID drive.
I get "LinkedObject: Link not restored" erros when loading the files and the paths are with the same issues, some parts are disappearing from the assembly.

I hate Python, but decided to dive into the code trying to find a solution.
It looks like the file paths are not handled in asm4 code, so its probably a code FreeCAD code.
@Zolko-123, can you point me to a proper place in FreeCAD code where I should start looking?

abetis, 2020-08-25:
I found this thread that reports exactly the same issue, but realthunder claims that it was fixed already...
https://forum.freecadweb.org/viewtopic.php?f=20&t=40403

I'm on 0.19 daily build, so I assume I was supposed to get that update...
Any ideas?
OS: Ubuntu 18.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Branch: unknown
Hash: 45d28c2367f5714cf5463b3373979c2a6f2779bf
Python version: 3.6.9
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

Zolko-123, 2020-08-25:
> I'm on 0.19 daily build

There are many problems with the Ubuntu daily-builds, could-you please try with the AppImage : 

https://github.com/FreeCAD/FreeCAD/releases/tag/0.19_pre

abetis, 2020-08-26:
Interesting... The icons in the application look different...

Anyway, running from the latest AppImage, did the same test in the NFS mounted directory...
Closed FreeCAD after creating all the files.
Run again, open the "base" file only, the tree is restored, but linked parts are red with exclamation mark and have those errors:
09:32:24  7.17e-07 <App> Document.cpp(2844): base#channel_1121_432mm.LinkedObject: Link not restored
Linked object: Model
Linked file: channel 1121 432mm.FCStd
09:32:24  9.1265e-05 <App> Document.cpp(2844): base#test_asm4.LinkedObject: Link not restored
Linked object: Model
Linked file: test_asm4.FCStd

The "File Name" of the linked parts goes down to the mount location instead of being relative.

In short, the same thing :(

I've searched for "Link not restored" message in github, but couldn't get to that code.
Do you know what repository has that code?

Zolko-123, 2020-09-02:
see https://forum.freecadweb.org/viewtopic.php?f=20&t=40403&p=427735#p427607

closed 2020-09-02 by Zolko-123

abetis, 2020-09-05:
@Zolko-123 
How can I know if the change is already in the daily build?
Thanks.