Zolko-123/FreeCAD_Assembly4 #100: Strange behavior of assembly with imported multi-part STEP

abetis, 2020-08-24:
Attaching the STEP and the FCStd.
I can use the part in other assembly, but there is something wrong with its behavior.
For example, I've added 4 hole axis to the assembly.
Those axis are not appearing as mounting points for the part when adding it to another assembly.
Tried to import datum as you proposed for other case, but get the "The selected datum object cannot be imported into this assembly" warning popup dialog and the task view is empty with only OK and Cancel buttons.
The report view shows "<App> Document.cpp(3407): Please reload partial document 'v-wheel set 24mm' for recomputation." warning.
[doc_335059_270531.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/5119600/doc_335059_270531.zip)
[v-wheel set 24mm.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/5119602/v-wheel.set.24mm.zip)

abetis, 2020-08-24:
As I understand, the file has ASM4 model inside, so I expected that I could link into it another object.
The link process succeed (to the default LCS since the holes axis are not visible), but changing the placement of the linked part does nothing, the linked part stays at the same location.
Attaching the FCStd of the part I've tried to link into the v-wheel assembly.
[spacer 6_8 4mm.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/5119619/spacer.6_8.4mm.zip)

Zolko-123, 2020-08-25:
> For example, I've added 4 hole axis to the assembly

how did you add them ? You seem to have copied the Part containing the STEP import into the assembly. There is nothing wrong with that, but in this case the included datums do not appear in the assembly as attachments

Zolko-123, 2020-08-25:
the proper way to do this is to copy the part containing all the STEP import into the Parts group, and link that part into your assembly. Please find attached your corrected file.

Actually, in this case, since there are 4 identical legs, you should only copy 1 of them and insert it 4 times. 
[v-wheel set 24mm_new.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/5122024/v-wheel.set.24mm_new.zip)
[v-wheel set 24mm_1.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/5122026/v-wheel.set.24mm_1.zip)

abetis, 2020-09-01:
Sorry, I don't follow how do you propose to do it...
I have another case, a thrust ball bearing.
I import it from STEP file that has multiple bodies grouped together.
The origin of the imported file is off the center of the bearing.
If the bearing was a single body object, I could click on the inner hole and add the hole axis, the hole axis would appear on the same level as the body in the assembly tree.
In case of grouped objects, the hole axis appear on the same level as the body I select, meaning it will be in the group, so that hole axis will not be visible when trying to attach that bearing in assembly.
Please see the attached files. Try to link the bearing to any other assembly.
How can I get the hole axis to be usable as an attachment point?
[bearing_step.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/5158202/bearing_step.zip)
[thrust ball bearing 8x16mm.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/5158208/thrust.ball.bearing.8x16mm.zip)

Zolko-123, 2020-09-07:
"Body" is a concept that belongs to the PartDesign WB, while "Part" is a general FreeCAD container. When importing STEP files, there shouldn't be any "Body" involved, you should have only "Solids" and "Part" containers. Could you please rephrase what you mean by using carefully the "Body" word ?

See in attached file what I would do.
[thrust ball bearing 8x16mm.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/5181849/thrust.ball.bearing.8x16mm.zip)

abetis, 2020-09-07:
FreeCADs terminology is pretty confusing and I'm not an advanced CAD user...
I'm trying to understand what you did in the file.
Since the imported STEP was multi-body/multi-solids (the blue icons), FC created part container (yellow icon) to group all the bodies together.
In that case, since its a part, you dragged it into "Parts" folder and created a link to that part?

Checked your tutorial1 again and I see that the bearing you import there wasn't grouped into part, it had only solids, so that's not a default FreeCAD behavior, but something that came from within the STEP file.

So if I drag all the solids into the "Model", and delete the part that came from the STEP file, I should basically have the same effect?

Zolko-123, 2020-09-07:
if you only want to use the imported STEP model, then there is no need to put it inside an Asm4 Model. If you have all the solids inside a Part container, you're good.

As for the tutorial, previously FreeCAD imported solids into the active document, and recently this behaviour changed and it puts the imported solids into a Part container. 

So, basically: create a new empty document, import the STEP file. If it creates a Part container, you're all set, if not, then create an empty Part container and drag all the solids into there.

abetis, 2020-09-07:
And if I want to add LCSs, such as hole axis?
Should I add the assembly structure to the document if the STEP is loaded with the part container?

Zolko-123, 2020-09-07:
in the **latest** Asm4 releases you don't need an Asm4 Model to do all this. You can add LCS and datums and everything to plain Part containers. At least you _should_ be able, if not it's a bug and please report.

abetis, 2020-09-07:
nice.
Just updated the plugin, will try and report if will face a problem.

closed 2020-11-20 by Zolko-123