Zolko-123/FreeCAD_Assembly4 #102: Advanced usage of variables

abetis, 2020-08-29:
Is there a documentation of the variables and formula editor besides the exmaple2 of asm4?

Specifically, I'd like to know if there is a way I can write a "function" that uses and changes the variable values?
3 use case examples:
1:2 gear attachment, I could add a "gear1angle" variable for one gear angle and would like to have a simple function that will set "gear2angle" to be half of gear1angle.

lead screw rotation moves the nut.
I could have a small function that changes the nutPosition variable according to leadScrewAngle (8mm for one rotation).
There will be more than one math operand, but could still be written as a single line math function.

Limit the movement of a part. Same lead screw example, but with a limit.
For example, the nut could move between ranges of 20-80mm.
If there was a way to have conditions in such function (IF-THEN-ELSE), I could check if the nutPosition is in the range, than change its position, if not, do not change it and somehow prevent the leadScrewAngle to change as well.

Is there a way to implement that?

Thanks.

Zolko-123, 2020-08-29:
yes, of course, that's the ExpressionEngine. You can invoke it either by clicking on the small f(x) icon at the right of the parameter field, or by beginning to enter "=" into the parameter field. 

For example, if you create 2 variables gearAngle1 and gearAngle2, you can set the ExpressionEngine in gearAngle2 to "=gearAngle1 * 1.5" and the second gear will rotate 1.5 times faster. You can do the same with a lead-screw, or any combination of fixed variable dependencies.

Zolko-123, 2020-08-29:
You have also conditional statements, but please look at:

https://wiki.freecadweb.org/Expressions

closed 2020-08-29 by Zolko-123

abetis, 2020-08-30:
So that's basically the same as spreadsheet and everything is limited to a small one-line input field...
There is no way to write a multi-line function.

Zolko-123, 2020-08-30:
Did you try with multiple intermediate variables to store intermediate results of your multiple lines ?

abetis, 2020-08-30:
Not yet.
I thought there might be a way to write a python-style function, which obviously will not be one line.
But if it's limited to a spreadsheet-style, it will be a bit harder for advanced things. 
Harder, but probably still possible.