Zolko-123/FreeCAD_Assembly4 #103: Usability: After a local coordinate system is created, we should enter "Edit Datum" directly

cheuksan, 2020-08-29:
Once I created a new local coordinate system, I always enter the "edit datum" dialog manually. It is also true with most of the assembly 4 tutorial videos on youtube. Can we go directly into "Edit Datum" without clicking after a new LCS is created?

https://www.youtube.com/channel/UCpdgfEnmKNi8iv_A66efg9Q/videos

[image]
[image]

Zolko-123, 2020-08-29:
I'll have to look

Zolko-123, 2020-08-31:
Done, will be in the next release

closed 2020-08-31 by Zolko-123