Zolko-123/FreeCAD_Assembly4 #105: [enhancement] Allow infinite animation

abetis, 2020-08-30:
Current animation allows only running from start to end values.
Sometimes I'd like to have an infinite animation for a presentation for example.

I can see 2 types of loops:
- For rotational animation I could set start as 0 and end as 360, so the application will just have to start again when it reaches the end value.
- For linear movement the application will have to negate the step and move from end to start when reaching end value and negate the step and and move again from start to end when reaching start value...

Zolko-123, 2020-08-31:
This is a very good idea. If you know Python and Qt, you can make the enhancement and propose it here.

abetis, 2020-08-31:
I hate python...
will check the code when I'll have a bit more time

abetis, 2020-08-31:
Related question:
Is there a way to set a variable value besides the animation window?

And the animation on my PC is very slow and jumpy, although I have a 12 core CPU and a good graphics card.
Is it related to assembly/animation or somewhere in the FreeCAD core?

Zolko-123, 2020-08-31:
You can set the values in the Property View window. Or Combo View. Select the Variables object in the tree, and in the Property View window you'll see all the variables and their values. 

Or do you mean something else ?

abetis, 2020-08-31:
Yes, that's exactly what I was looking for.
Don't know if it's related to the delays on my PC, but I see that sometimes the parts that are bound to a variable are not calculated when animated, but do calculate when manually change them.

For example, I have 2 sliders on a channel, for one of them the Z axis is bound to variable "nutPosition", for the other the axis is "14-nutPosition", so one goes in positive direction and another in negative direction.
I was trying to find the needed constant (14) to set the second slider to the proper place to have symmetric movement.
With the animation window, while manually changing the slider position I got to value 20, when changing the variable manually via the property window I got to value 14, while the 3D view changed on every change of the variable value.
Looks like not every change of the animation window updates the 3D view. Or not all the objects in the view are updated...
Maybe I'll have more understanding of what is going on there when I'll look on the code.

abetis, 2020-08-31:
Actually... The manual variable update doesn't work well too.
I set a value, see the part move.
Click on the value again in the property view, type in exactly the same value, click somewhere else to loose the focus (or press TAB) and the part move again.
Pressing recalculate between the changes doesn't help.

Zolko-123, 2020-09-02:
done, will be in the next release

abetis, 2020-09-02:
What did you change?

Zolko-123, 2020-09-02:
is in v0.9.8

closed 2020-09-02 by Zolko-123