Zolko-123/FreeCAD_Assembly4 #108: Variables of sub-assembles

abetis, 2020-09-05:
I have an assembly of sub-assemblies.
Sub-assembles have variables that affect its appearance and can be used for animation.

Those variables are not visible on the father assembly level.

Is there any solution for that?

Zolko-123, 2020-09-07:
you can read them in the parent assembly, but you cannot set them there. To read a variable from a sub-assembly, create another variable in the parent assembly (with the same name or not) and set the value of the variable to subasmname.Variable.varname (or similar, you must check).

If you want to set a variable in both assembly and sub-assembly, you can create a master-document, set the variable there, and insert (link) that master-document into the assembly and sub-assembly.

I've not tested this very much, please report here what works and what doesn't.

abetis, 2020-09-07:
Will try when I'll get home.
The unability to set the variables of the sub-assembly breaks the animation on top-level assembly :(

abetis, 2020-09-07:
Follow up question, that might answer another question:
I'm (slowly) reading the long assembly4 forum thread and someone mentioned there a "setup" of values.
Like a "closed" position for a piston and "open" position.

Do you see a way to implement it somehow on the way?
Let's say I add a spreadsheet to the project, have there a table of variables (rows) and setup options (columns).
Assuming I fill those values manually in the spreadsheet...
Add that spreadsheet to all sub-assemblies if needed.

Can python script go over the variables and set the specific values into the variables on the sub-assembly level? Or sub-sub-assembly level?
Assuming the spreadsheet will have the "hints" for the script with the proper naming that should be used to set the values (if that's possible at all).

Zolko-123, 2020-09-07:
re-setup values: you can do it for a child object and all its instances, but not on a per-instance of a child. This is a concept we have called "variant link" and is not yet implemented.

Zolko-123, 2021-10-04:
variant links are implemented in v0.11, see:

https://forum.freecadweb.org/viewtopic.php?f=24&t=62558

closed 2021-10-04 by Zolko-123