Zolko-123/FreeCAD_Assembly4 #109: Code review: redraw speed

abetis, 2020-09-06:
Hi @Zolko-123,

Probably not an issue, but more as a code review that I wanted to do for drawing speed of assemblies.
That's probably a better place than a new forum thread (or the long assembly4 thread) since it's more related to code and can point on some places in github editor.

I face slow redraws of my assembly, which is probably not small, but also not so large that should slow down the things so much.
As I understand, the whole asm4 idea was to get around the asm3 resolver and speed up the assembly time since all the attachments are mathematically bound.

Assuming there is no animation involved, the assembly is static.
Zooming or rotating the screen is very slow.
Can you think of a function in asm4 that is called/polled during the screen redraw that might slow the things down?

Some information:
- All my objects are imported from STEP files, that might include some complex bodies, such as threads, gear teeths and so on...
- I'm currently running on locally compiled version with default settings - can it slow down things?

Thanks.

Zolko-123, 2020-09-07:
Yes, I've seen that too.

Nominally, during screen redraws (pure 3D movements) there shouldn't be any recalculations, therefore I don't understand the slowness either. 

Do you have the possibility to export your full 3D assembly to STEP, and re-import it into FreeCAD, and see whether this static version of your assembly has the same issue ?

abetis, 2020-09-07:
Will try that later.
The feeling is as there is a memory leak or a very lists that should be constantly processed.
I had situations that not only the 3D view become slow, but also had to wait until tree view responded to clicks and scroll...
Is there a way to debug it somehow?

abetis, 2020-09-07:
Something doesn't work with step file export for pretty simple parts :(
Can you try exporting the attached file to STEP?
[channel 1121 192mm.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/5184494/channel.1121.192mm.zip)
On my side the STEP is created, about 1Mb, but on export I get an error popup saying that it can't load the STEP (don't know why it's trying to load it if I want to export)...
Update: I managed to load the created STEP, but the view is some how clipped and I can see the part only when rotating it to some specific angle.

I've tried to export the lagging assembly, which created about 200M file, that I gave up loading after about 15 minutes...
Don't really know how to debug it

abetis, 2020-09-07:
Can't find the file of the lego garden with 300 parts that you posted in the forum.
Can you send it please?
Want to know if the problem is with the parts complex shapes or the number of parts.
The lego garden parts are more simpler shape I think.

closed 2021-10-04 by Zolko-123