Zolko-123/FreeCAD_Assembly4 #111: Selecting a part in the 3D view should select asm4 level entry (link) instead of the object body

abetis, 2020-09-06:
With large assemblies, when the tree view is very long, it's easier to select the parts by clicking on them in the 3D view.
Lets say I want to change the parts attachment position.
When you click on a part, the link entry in the tree view is expanded and the inner body is selected instead of the asm4 link entry.
To change the attachment position I have to click on the asm4 level entry in the tree to get to the property.
The situation even worse when you have few levels of sub-assemblies involved and click on a part from a sub-sub-assembly that unwraps many-many levels in the tree.

Zolko-123, 2020-09-07:
Yes, you're right, this should be the better behaviour, but I'm not sure how to do it. I'll try to come up with something.

abetis, 2020-09-10:
Please see #113

Zolko-123, 2020-11-15:
I believe this is fixed with #113

closed 2020-11-15 by Zolko-123