Zolko-123/FreeCAD_Assembly4 #118: MakeBOM should create a spreadsheet so that it can be inserted into a drawing

xoviat, 2020-09-18:
If this idea is accepted, then I may implement it.

Zolko-123, 2020-09-21:
yes, this is the planned behaviour. It would be very helpful if you could propose an implementation

xoviat, 2020-09-22:
This is the macro that I'm currently using (I've written in within the last few days). It creates a BOM spreadsheet in the model and then inserts it into the drawing. The titles are taken from the part label as follows:

[image]

[image]

(Forgive these titles; this assembly is WIP) 

```
import FreeCAD
import Draft
import datetime

from fractions import Fraction
from typing import List

def Partlist(object,level=0,tab=None):
	forbbox = ('PartDesign::Body', 'Part::Feature', 'Part::FeaturePython')
	indent = '  '*level
	if object == None:
		return {}
	if tab == None:
		tab = {}
	if object.TypeId=='App::Link':
		print(indent+object.Label+' -> '+object.LinkedObject.Document.Name+'#'+object.LinkedObject.Label+' => '+object.LinkedObject.FullName)
		Partlist(object.LinkedObject,level+1,tab)
	else:
		print(indent+object.Label+' ('+object.TypeId+')')
		if object.TypeId=='App::Part':
			if level > 0:
				if object.FullName in tab:
					tab[object.FullName]['Qty.'] = tab[object.FullName]['Qty.'] + 1
				else:
					tab[object.FullName] = {}
					tab[object.FullName]["var"] = {} 
					tab[object.FullName]['Qty.'] = 1
				tab[object.FullName]['Ref No.'] = object.Document.Label
				tab[object.FullName]['Title'] = object.Label

			# look for Variables
			if object.Document.getObject( 'Variables' ):
				print(indent+'  Variables:')
				vars = object.Document.getObject( 'Variables' )
				for prop in vars.PropertiesList:
					if vars.getGroupOfProperty(prop)=='Variables' :
						propValue = vars.getPropertyByName(prop)
						print(indent+'	'+prop+' = '+str(propValue))
						if not object.FullName in tab:
							tab[object.FullName] = {}
							tab[object.FullName]['fullname'] = object.FullName
							tab[object.FullName]["var"] = {}
						tab[object.FullName]["var"][prop] = str(propValue) 
			# look for sub-objects
			for objname in object.getSubObjects():
				subobj = object.Document.getObject( objname[0:-1] )
				Partlist(subobj,level+1, tab)
	return tab

def CreateBOM(model):
	document = FreeCAD.ActiveDocument
	plist = Partlist(model)
	if len(plist) == 0:
		return

	if not hasattr(document, 'Spreadsheet'):
		spreadsheet = document.addObject('Spreadsheet::Sheet','Spreadsheet')
	else:
		spreadsheet = document.Spreadsheet

	spreadsheet.setColumnWidth('B', 350)
	spreadsheet.Label = "BOM"

	def wrow(drow: List[str], row: int):
		for i, d in enumerate(drow):
			spreadsheet.set(str(chr(ord('a') + i)).upper()+str(row+1), d)

	wrow(['Ref No.', 'Title', 'Qty.'], 0)
	for i, (_, data) in enumerate(plist.items(), start=1):
		wrow([data['Ref No.'], data['Title'], '{:.0f}'.format(data['Qty.'])], i)
	
	if not hasattr(document, 'Sheet'):
		sheet = document.addObject('TechDraw::DrawViewSpreadsheet','Sheet')
		sheet.Source = document.Spreadsheet
		sheet.CellEnd = "C" + '{:.0f}'.format(len(plist)+1)
		sheet.Label = ""

		document.Page.addView(sheet)

def main():
	document = FreeCAD.ActiveDocument
	model = document.getObject("Model")
	if model == None:
		model = document.getObject("Part")			

	CreateBOM(model)

	document.recompute()

if __name__ == '__main__':
	main()
```

xoviat, 2020-09-22:
On second thought, I may not want to list the parts in the subassemblies in the BOM.  I generally should only list only the subassemblies themselves.

Cyclonit, 2020-09-22:
A toggle to decide if/which subassemblies are to be included would be ideal.

Zolko-123, 2020-09-23:
Nice, thank-you. Some comments:

- you create the BOM by calling the spreadsheet "Spreadsheet", shouldn't this be called "BoM" ?
`BoM = document.addObject('Spreadsheet::Sheet','BillOfMaterials')`

- this is a macro, ideally this should be a command: did you have a look at how Asm4 commands are done, and how to adapt the macro to make a command out of it ?

- there is a function in libAsm4 that does already this:
`model = document.getObject("Model")`
`model = Asm4.checkModel()`
which returns `None` if no Asm4 Model is present. 

- alternatively, we can create a BoM on any selected App::Part container, which could also be useful.

xoviat, 2020-10-03:
Zolko, I can work on a PR so that we can discuss the implementation. However, I thought about replacing the existing BoM button. Would that be acceptable to you?

Zolko-123, 2020-10-04:
The current BoM button was copied from A2+, so please feel free to make your own.

Did you see the "Edit Part Information" button ? The purpose of this is to add user-provided information to a part, and some of that information should also be displayed in the BoM (like provider, #ref ...). The fields are currently hard-coded (and come from A2+ too) but it might make sense to be able to add custom fields. Ideally be able to load from templates, thus a designer can make his Part-Info template, and the relevant fields would be included into the BoM. These fields should only be informative, not technical (like material, density....)

Does that make sense to you ?

Cyclonit, 2020-10-15:
Why shouldn't technical properties be included? I'd expect a BoM to contain lengths of stock, screw diameters etc.

Zolko-123, 2020-10-15:
Yes, you're actually right. Ideally, if it's a spreadsheet, it should contain all available information, and then the user can delete unwanted columns.

DerAndere1, 2020-11-11:
I think a Build-in BoM should not deal with Metadata such as Screw diameter which is Not meaningful for Most Parts. For such Purpose, The User better creates a Relational database outside of the CAD program that connects Part ID  with the 3D Model and Child-Part IDs, Child-Part quantity and metadata (Description, Storage Location, Supplier, quantity in Stock).

FarmingSoul, 2020-11-12:
hello,
I am Kevin from FarmingSoul (https://www.FarmingSoul.org)
I work on CAD to make open tools drawings what be built by agricultor during a formation or them self.
FarmingSoul cooperate with l'Atelier Paysan (https://latelierpaysan.org) for lot of things.
we have a large data base of drawings , but we work with SW. (https://latelierpaysan.org/Plans-et-Tutoriels)
We try to migrate to FreeCAD (I work on it).
In first i worked with AS2+ because it can make BOM, but for some cylinder problems and others i prefered AS4.
So i work on the possibilities to have BOM.
But a special BOM because after we have a automatisation of every things (ordering, cutting sheet ...)
The description of part must be the same what we have in data base.
So i make a script for normal section bar, screw... but we have some special parts (motor, bearings , laser cutting ...)
and put manualy the informations is the unique way for there.

So i think the top is a "Edit part info" with button for script for 'normals' parts
and a BOM in Spreadsheet.

it's very dirt but i make that.
the script for 'défault part' work actualy only with my way to do the parts.
try this

[image]
[image]
[image]
[image]

[Assembly4.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/5528300/Assembly4.zip)

Zolko-123, 2020-11-14:
nice project, I check the BoM

cooltehno, 2021-08-19:
Hello, @FarmingSoul!
Your tool is looking great! Could you clarify for a simple user how to install your script? Which build of freecad should I use? Currently I have an official FreeCad 0.19 on Ubuntu 20.04.2.
Big thanks!

cooltehno, 2021-08-29:
Hello again, @FarmingSoul!
It looks like I figured it out :) Thank you for detailed explanation! I had no idea that your script can even build names (Reference_AP) - super!!
Also I've understood, that the "longueur" depends only on Pad's length (no matter how this defined through the Spreadsheet or directly).
With the reason of some language barrier and some inattention I mixed up "largeur" and "longueur".. I thought If I call any dimension in Sketch or  Pad's Length   "largeur" through the Spreadsheet - the script will take this as "longueur" and put it into BOM:))

As about my production I make some bar counters, kitchen furniture where I usually use metal frames covered by different sheet materials. I use Blender 3D for design. Blender has some capabilities for dimensions, but not so developed as FreeCad.

Now with your script I can easily make 1D cut-lists for lined tubes (I can insert Lengths and Quantity column to optimization program and get the cutting map immediately)!!

Still, something is missing for my needs.. For example rounded tubes (I mean arcs). Here I've made an arced tube:
[image]
I've made a body with "Sweep along a Pass" tool to have a possibility to set the length of the arc in millimeters (not angle, through a spreadsheet). Of coarse I'd be happy to use this dimension in my BOMs also. I can assign this number manually in the "Edit Part Information (FS Mod)" dialog, but automatic function could be incredibly cool!

For sheet materials it would be great to add some column "Width" ("largeur" if I'm not mistaken :) ) straight behind the "longueur" column. This could let to make some 2D cut-lists in optimization cutting programs.

BIG, BIG thanks!! Good Luck!

cooltehno, 2021-08-29:
An example of my test work in Asm4:
[image]

FarmingSoul, 2021-08-30:
hello @cooltehno ,

thanks for the test !

can you give me 2 files in Asm4 :
* 1 with your body with "Sweep along a Pass" tool
* 1 with your sheet materials
and can you give me your preference for the title of column
and your favorite translation for that :
"info par default"  :
"fer rond" + diametre for juste the constraints diametre
"fer plat" + largeur + X + hauteur for the constraints largeur, epaisseur, conge
"étiré plat" + largeur + X + hauteur for the constraints largeur, epaisseur
"tube rond" + diametre + X + epaisseur for the constraints diametre, epaisseur
"tube carré " + largeur + X + epaisseur for the constraints largeur, hauteur, epaisseur (for largeur = hauteur)
"tube rectangulaire" + largeur + X + hauteur + X + epaisseur for the constraints largeur , hauteur, epaisseur ( without largeur = hauteur )
and the name for your sheet materials.

if not, what program do you use for optimization cutting 1 and 2 D ?

Have nice day

Zolko-123, 2021-08-30:
A few comments: 

- can you please separate PRs for **BoM** and for **PartInfo** please ?
- please don't modify *Asm4_libs* for the default **PartInfo** fields. Fields should be added and deleted through the _PartInfo_ UI
- ideally, we would want some sort of templates to create Parts with default fields adjusted by the user. I don't know how to do that unfortunately
- Parts can be either App::Part (Std_Part) objects or PartDesign::Body objects.

FarmingSoul, 2021-09-01:
hello @Zolko-123 ,

* yes i will separate PRs.
* i think the default PartInfo fields is personnal for each user not for each assembly .
so the user must can be have this own fixe configuration (and the update of Asm4 must not destroy them)
what you think about go out this code part in external file write only at the first use and modifiable by the user

in infoPartCmd.py
```
 import os
 ### to have the dir of external configuration file
 wbPath = os.path.dirname(__file__)
 InfoKeysFile = wbPath + '\\InfoKeys3.py'
 
 
 from PySide import QtGui, QtCore
 import FreeCADGui as Gui
 import FreeCAD as App
 from FreeCAD import Console as FCC
 
 import Asm4_libs as Asm4
 ### try to open existing external configuration file of user
 try :
      fichier = open(InfoKeysFile, 'r')
      fichier.close()
      import InfoKeys3 as InfoKeys
      ### after we can replace Asm4.partInfo by InfoKeys.partInfo
      fichier.close()
 ### else make the default external configuration file
 except :
      fichier = open(InfoKeysFile, 'x')
      fichier = open(InfoKeysFile, 'w+')
      fichier.writelines("#!/usr/bin/env python3\n# coding: utf-8\n#\n# LGPL\n# Copyright HUBERT Zoltán\n#\n# infoKeys.py\n\n\n#customization file for the description of the parts in BoM list\n\n\npartInfo = ['PartDescription', 'PartID', 'PartName', 'PartSupplier']")
      fichier.close()
      import InfoKeys3 as InfoKeys
      ### after we can replace Asm4.partInfo by InfoKeys.partInfo

```
after we will have the right code to modify this through the PartInfo UI.

>ideally, we would want some sort of templates to create Parts with default fields adjusted by the user. I don't know how to do that unfortunately

* i gone make that after because i think the same  ;-)

xoviat, 2021-09-10:
One thing I like about the macro that I've been using is that it avoids duplicate information. The way I have the macro set-up is that it sets the drawing title to the label that's given to the part. Therefore, the label drives both the name in the parts list and the drawing title name:

[image]

[image]

One thing I actually like about FreeCAD better than Inventor is that the title needs to be set in multiple places in Inventor, whereas in FreeCAD I only need to set the title in one location. I am somewhat against adding a 'Parts Info' dialog for that reason, specifically where such dialog stores the drawing title in some other location than the Label, as it would lead to duplicate information. I also like the fact that the drawing number is also stored in one location, namely the file name. In the scheme I am using, there is no location where the drawing number is stored other than the file name (the macro simply sets the drawing number field to the file name). This, however, may be a step too far, because others may want to change their filenames without changing the drawing number.

I think that if you're an organization or individual who is going to be creating 'real' designs, every drawing needs to have a number and it needs to be stored in the same location (whether it's an assembly or part). Therefore, people can access the drawings by simply referencing the number.  When I create a new drawing I simply give it the lowest unused number (next on the list). Each drawing number is then going to have a drawing title (I'm not particularly good with titles, but I try). Each drawing MAY have an item number (the drawing number is unique to me and identifies the drawings that I create, but the item number may reference someone else's part number or such). Giving your drawings special names only works if you only have a few projects, but once you have hundreds or thousands of drawings, you need a more standardized system. 

Anyway, those are just my thoughts. I don't necessarily want to block a useful feature, but I want to offer my feedback.

FarmingSoul, 2021-09-15:
Hello @xoviat ,@Zolko-123 ,

I make a draw for explain myself.

Here we see roughly the purpose of using FreeCAD for me, and I think others like @cooltechno, and other manufacturers that use products sold by linear or square meters and manufactured products.

[image]

hoping that this graph helps us to understand each other

Bye everybody,
Have nice day

cooltehno, 2021-09-16:
Hi, @Zolko-123 @FarmingSoul & everybody!
I've made git account under the same name, as I have in FreeCad forum - "totalcad", to avoid jumble..)) I'll continue the discussion under "totalcad" nick here

totalcad, 2021-09-16:
It's new me!(cooltehno)
@FarmingSoul ! Thank you, for a cool expansive job!
Here's my example files (in .zip):
 - arc_tube_with_cuts_from_sketch - "1 with your body with "Sweep along a Pass" tool";
 - arc_tube_with_cuts_from_binder - a tube made from complex sketch using ''Binder' tool (useful when you have many tubes connected with many different angles)(I plan to upload a complex garden furniture bench assembly using this technique);
 - linear_tube - a simple Pad-based body;
 - linear_tube_with_cuts - the same as previous but two angle cuts added;
 - wooden_plate - an example of sheet material with typical AxB (Length x Width) dimentions;
 - concrete_block - simple Pad-based body with volume formula in the Spreadsheet and Lable.

All parts are collected in general assembly "ASSEMBLY_FOR_BOM_EXAMPLE" - here's a spreadsheet BOM generated with ASM4 BOM tool (amazing "(FS Mod)"). After generating I've corrected BOM in the way of my vision:
Just simple columns: Qnt.; Name; dim1; dim2; dim;3.... or with simple specialized labels: Qnt.; Name; length, mm; width,mm, volume, m^3; mass, kg; info...

[bom_request_example.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/7174011/bom_request_example.zip)

I've taken "mass" data from "FCInfo" script: I've chosen every part one by one and copied manually mass data (entered density for steel - 7,870kg/dm3; for concrete - 2.5kg/dm3; for wood - 0.7kg/dm3) into Edit Part Information dialog.
[image]
All the parts have a unique color - which I've tried to visualize in the BOM.

Ideally, I'd be super happy if Edit Part Information dialog could have possibility to enter a formula, like almost every parameter in FC has!))) But I can understand how much effort this can consume!! - anyway it's just my wish :)
[image]

totalcad, 2021-09-16:
@FarmingSoul !

Here's my optimizing kit of programs you've asked:

https://www.nesting-technologies.com/realcut1dx-en.html - for linear optimization;

https://www.nesting-technologies.com/simple-cutting-software-x-en.html - for two dimensions cutting

My personal using experience with your's fantastic FS Mod:
https://disk.yandex.ru/i/1EqQAFAfD2vydQ

8 garden furniture benches assembly itself:
https://disk.yandex.ru/d/kHzbpf11GdKM8g

totalcad, 2021-09-16:
@FarmingSoul, could you make an English version of you "manufacture map"? :)
Thanks!

FarmingSoul, 2021-09-16:
Hello,
I update  "manufacture map" in english .

FarmingSoul, 2021-09-16:
Hello @totalcad ,

I see your video :
>My personal using experience with your's fantastic FS Mod:
>https://disk.yandex.ru/i/1EqQAFAfD2vydQ

And we work with more or less the same method.

I show your work and i return after.

Nice day

Zolko-123, 2021-09-16:
Hello All,

It's great to see your advances on this PartInfo and BoM functionality. I think this is really a very important thing for real-life production.

Unfortunately I didn't follow very closely your development (many other things to do, sorry), so could you please give a short summary of where we stand, and the next steps that should be taken ? I feel that PartInfo should be improved, what do you think ?

totalcad, 2021-09-17:
@FarmingSoul, I've forgot here's the link to the forum:
https://forum.freecadweb.org/viewtopic.php?f=20&t=34806&start=960
Additional videos (may be could have some clarification):
https://disk.yandex.ru/i/D3SU2-98YpsB1Q - a structure of the main assembly
https://disk.yandex.ru/i/mzcSVRySKuqI4Q - sheet metal detail video

Good day!

totalcad, 2021-09-21:
Hello, @FarmingSoul, sorry for a cascade of request! :))...
One more the parameter for calculations in BOM. If we need to calculate some cnc/laser cut length -we could measure&use the "Perimeter of a Sketch" - i've done a topic in FC forum here:
https://forum.freecadweb.org/viewtopic.php?f=8&t=62391

[image]
If you'll need the FC Exrpession  - people answered me: "Sketch.Shape.Length" - this works for formulas - I've tested!
Thanks&good day!

FarmingSoul, 2021-09-28:
hello @xoviat ,
Can you test this PR #246  .
It 's make 'for you' (because i just put in autofield info "name of doc","name of part" ).
Thanks

FarmingSoul, 2021-09-29:
hello @totalcad ,
sorry I put our work totally aside to try to do something totally configurable.
I think soon we will close this issue and maybe reopen another one to discuss useful and self-fillable default fields.
you can get ahead of the curve by downloading this #246 .

totalcad, 2021-09-29:
@FarmingSoul thanks for sharing & good dialog!
I'm already glad to use your tool as it is! Good mode & luck!

Zolko-123, 2021-10-04:
see PR https://github.com/Zolko-123/FreeCAD_Assembly4/pull/253

closed 2021-10-04 by Zolko-123

markbil, 2021-11-22:
hi all. thanks for the great work on assembly4. I am designing my kitchen cabinets in freecad, and since I updated assembly4 to version 0.11.4 from an older version through the addon-manager, the BOM feature (Menu => Assembly => Create Part List) does not seem to work any longer. in the older version a popup window would appear with a list of all my Part names (kitchen cabinet panels) and their dimensions, i.e. something similar to "Top Panel: 600 x 18 x 450, etc.". Now when I execute the same function I receive an empty popup as pasted below... Also the report view throws "10:28:40  3634.66 Selection.cpp(1649): Object not found". I changed nothing but updading assembly4 to the latest version...
Am I doing something wrong? Thanks for your help!
[image]

snelweg, 2022-01-11:
> I changed nothing but updading assembly4 to the latest version... 
[image]

Same here

```
10:22:16  Initializing Assembly4 workbench (v0.11.4) ............................ done.
10:22:18  Activating Assembly4 WorkBench
10:22:33  1.664e-06 Selection.cpp(1649): Object not found

```

> OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/cinnamon)
> Word size of OS: 64-bit
> Word size of FreeCAD: 64-bit
> Version: 0.19.24366 (Git) AppImage
> Build type: Release
> Branch: (HEAD detached at 0f9259c)
> Hash: 0f9259cda103ae1824ac16c68ac9b4a0d54b05fc
> Python version: 3.9.7
> Qt version: 5.12.9
> Coin version: 4.0.0
> OCC version: 7.5.3
> Locale: English/United States (en_US)

robertlinn0423, 2022-04-23:
I also see the empty popup window and receive the 'Selection.cpp(1649): Object not found'  error:

OS: Ubuntu 21.10 (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24366 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0f9259c)
Hash: 0f9259cda103ae1824ac16c68ac9b4a0d54b05fc
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: English/United States (en_US)

robertlinn0423, 2022-04-23:
Success!!! I was able to create the Parts List.
After reviewing [bom_request_example.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/7174011/bom_request_example.zip) provided by totalcad on Sept. 14, 2021 here's the steps I used.

=== Create the Part(s) ===
# Create a new project for the Part: File/New
# Switch to Assembly 4 workbench.
## Create a new Part and select it.
## Create a new Body under the Part and select it.
## Right mouse button on the Body and click 'Toggle Active Body'.  This will switch to Part Design.
## Now in 'Part Design Workbench' create a new Sketch, constrain and pad it to create a solid. (Repeat this for each part)
## Save the project.
## Close the project and re-open it.
=== Create the Assembly ===
# Switch to Assembly 4 workbench.
## Create a new project for the Assembly.
## Create a new Assembly Container.  Make sure it's the active document.
## Save the Assembly project.
## Use the 'Insert Link to a Part' and link the Part(s) previously created.
## Save the Assembly project.
## Select the Assembly Project
## Click the 'Parts List' button.  This opens a window indicating the BOM Spreadsheet was created.
## Open the newly created BOM Spreadsheet to review the parts.