Zolko-123/FreeCAD_Assembly4 #128: Strange behavior when moving LCS from Model(assembly) into Part

AcumenDev, 2020-10-03:
When I move LCS from Model(assembly) into Part this operation destroys the object Boby

1)Example create one part and assembly model 
[image]
2)I created LCS in the wrong place 
[image]
3)now I'm moving an LCS to Part. This operation broke the Body(steps fo create body now in body level)
[image]
[image]

Thank you for Assembly 4

FreeCAD Assembly 4 workbench
Current version 0.9.9, 2020-09-24

OS: Debian GNU/Linux 10 (buster) (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22522 (Git) AppImage
Build type: Release
Branch: master
Hash: d8e476ef428017900dfe0f9359ba448503c216f9
Python version: 3.8.5
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

Zolko-123, 2020-10-04:
> _I created LCS in the wrong place_

yes, this happens with me also sometimes, but then I delete the LCS and create it in the correct place.

closed 2020-10-04 by Zolko-123

AcumenDev, 2020-10-04:
> > _I created LCS in the wrong place_
> 
> yes, this happens with me also sometimes, but then I delete the LCS and create it in the correct place.

This behavior Freecad itself or assembly 4?
There is another question: where the correct place do create LCS in the part or in the body?
May be correct to create LCS in an active node(part for example) in the project tree, not in a dedicated node (Because of these behaviors LCS are often created in the wrong place namely in the Assembly model node.)?