Zolko-123/FreeCAD_Assembly4 #130: International symbols not shown correctly in Body names created within A4 workbench

timtashpulatov, 2020-10-05:
If a new Body is created in Assembly 4 workbench, and the new Body's name contains international characters (Cyrillic for example), these symbols are shown as underscores.

[image]

When a Body is created using Part Design workbench, international symbols are shown correctly.

Zolko-123, 2020-10-05:
> _When a Body is created using Part Design workbench, international symbols are shown  correctly_.

When a Body is created using PartDesign workbench, it is called "Body" therefore its name doesn't contain any "international" characters. You're confusing "Name" and "Label"

closed 2020-10-05 by Zolko-123

timtashpulatov, 2020-10-05:
> When a Body is created using PartDesign workbench, it is called "Body" therefore its name doesn't contain any "international" characters. You're confusing "Name" and "Label"

Most likely I am confusing Name and Label, indeed. Still could something be done to avoid those underscores when using international characters? Thanks.

Zolko-123, 2020-10-05:
did you try to rename the Body that you created ?

timtashpulatov, 2020-10-05:
Renaming the Body seems to work. Underscores are still shown in Part List when inserting link to a part, but I guess I can live with it.

1. Body created with int'l name:
[image]

2. Body renamed:
[image]