Zolko-123/FreeCAD_Assembly4 #146: adding a model cannot be undone

donovaly, 2020-11-22:
when using the menu Assembly -> New model, this action cannot be undone.
Take for example this file having 2 parts:
[Cube-Test.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/5580267/Cube-Test.zip)

When the new model is created the bodies are automatically put into a group called "parts". When you now realize that you e.g. don't need the model, you cannot simply press undo.

I have Assembly 0.9.12

Zolko-123, 2020-11-22:
you're right.

closed 2020-11-22 by Zolko-123

donovaly, 2020-11-22:
Why do you close this? You verified the bug is there, so let's add the undo registering, no?

Zolko-123, 2020-11-22:
On dimanche 22 novembre 2020 21:54:05 CET Uwe wrote:
> Why do you close this? You verified the bug is there, so let's add the undo registering, no?

It's not a bug. If you want this feature, please propose a pull request.

Zoltan

-- 

═══════════════
   HUBERT Zoltán 
═══════════════

donovaly, 2020-11-22:
Why is this not a bug? I perform an assembly 4 action and this should be undoable like any other action invoked by assembly 4.

Zolko-123, 2020-11-23:
On dimanche 22 novembre 2020 22:17:09 CET Uwe wrote:
> Why is this not a bug?

because it's a feature request. I'll review your PR with great care.

z

donovaly, 2020-11-23:
> because it's a feature request

Please don't play Ping Pong.
The action is Assembly 4 only. There is no bug in FC since there all additions can be undone.
Here the model addition is done by Assembly 4, so Assembly 4 is responsible to register its undo. This is not the case but it must do so according to the guideline for FC workbenches. So it is a bug in Assembly 4.

It is not that pressing, but please keep the bug report open that it is not forgotten. Thanks.

If you insist it is not a bug please explain why you think so.

Zolko-123, 2020-11-23:
On lundi 23 novembre 2020 11:03:50 CET Uwe wrote:
> > because it's a feature request
>
> If you insist it is not a bug please explain why you think so.

you said it yourself:

On dimanche 22 novembre 2020 21:54:05 CET Uwe wrote:
> so let's *add* the undo registering, no?

yes, "let's" .  As I said, please submit your PR and I'll review it with great attention.

z

-- 

═══════════════
   HUBERT Zoltán 
═══════════════

donovaly, 2020-11-23:
> so let's *add* the undo registering, no?

Yeas to fix the bug.

> As I said, please submit your PR and I'll review it with great attention

Is this your way of developing - if one cannot fix a bug immediately, one is not allowed to open a bug ticket?
Like with the FC issue tracker, I collect bugs I stumble over. Since I develop in my spare time, I will have a look when I find time.

So please reopen this ticket that we have a reference to this issue, that others don't report the same thing and that someone who finds time will one time fix it.

Zolko-123, 2020-11-23:
What is it that you don't understand ? 

z

On lundi 23 novembre 2020 14:08:28 CET Uwe wrote:
> Yeas to fix the bug.
>
>
> Is this your way of developing - if one cannot fix a bug immediately, one is not allowed to open a bug ticket?
> Like with the FC issue tracker, I collect bugs I stumble over. [...]

-- 

═══════════════
   HUBERT Zoltán 
═══════════════

donovaly, 2020-11-23:
> What is it that you don't understand ?

I think I made clear that this here is a bug tracker to collect issues, no matter if one can fix them immediately or some months later. I also made clear that it is an issue of assembly 4, not FreeCAD, so the issue should be recorded here.
But you immediately close it that contradicts the idea of a bug tracker.