Zolko-123/FreeCAD_Assembly4 #150: TechDraw view fails with assembly

everlanes, 2020-11-30:
When I tried to add an assembly to a TechDraw View the view was empty and I got this error:
 GO::projectShape - OCC error - BRep_API: command not done - while extracting edges

Digging deeper, I found, that it only happens, when the assembly has a visible Axis Datum. Removing the Axis Datum or hiding it, fixes the problem.

I guess TechDraw tries to draw the Axis Datum and fails ...

See also the attached minimal example file:
[Test_TechDraw.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/5616218/Test_TechDraw.zip)

Zolko-123, 2020-11-30:
1) a FreeCAD file is already zipped, it's useless to zip it again

2) this is actually a TechDraw bug: when a datum line is inside a body, it can handle the view correctly (axis is not displayed) but if you drag the datum line outside the body, but still in the part, it throws an error.

everlanes, 2020-11-30:
1. Github doesn't accept *.FCStd files as attachment - easiest to zip it again to get correct extension
2. No, the datum line is in the center of a cylinder and still not working when visible.

I know, that it is easy to blame TechDraw, but Assembly 4 is the new kid in town and this error happens with Assembly 4. I don't know, if blaming other workbench is the best way to go from here.

Just my 2 cent

Zolko-123, 2020-11-30:
try this, without using Assembly 4: create a Part, inside the Part create a Body, there create a circle, extrude it (to a cylinder), make a TechDraw of the Part. 

Then, create a Datum line in the Body: TechDraw view updates correctly.
Then drag the datum line outside the Body but still in the Part: TechDraw makes that error.

`21:39:03  GO::projectShape - OCC error - BRep_API: command not done - while extracting edges`

No Assembly 4 involved. But it's a good catch, you should report the bug to TechDraw.

closed 2020-11-30 by Zolko-123

everlanes, 2020-12-01:
Ok, thanks for the insight. I can reproduce it now without Assembly 4 and I have reported it in the FreeCAD forum.

aapo-aapo, 2020-12-04:
It is indeed a TechDraw bug.  I made a draft [Pull Request](https://github.com/FreeCAD/FreeCAD/pull/4106), which should fix the issue, but due to the rather extensive changes in a complicated c++ code using a smorgasbord of links and documentobjects I'm not overly confident that the fix doesn't have other unfortunate side effects.  But hopefully, a fix is on the way!