Zolko-123/FreeCAD_Assembly4 #161: Animation dialog does not correctly store previous state

donovaly, 2021-01-11:
I don't know if this is a bug or missing feature, however, it would be very helpful if the animation dialog would state the previous state since at the moment you have to reset the value range on every call of the dialog.

Here is an example:
[image]

So you can see the float ran from 0 to 360 but after the reopening there is no range anymore. It stored the last value and put this to both range fields.
However, the state of the Loop option is correctly stored.

Zolko-123, 2021-01-12:
This has been asked before, I have to think how to implement it (a global variable may-be)

donovaly, 2021-01-12:
> This has been asked before, I have to think how to implement it (a global variable may-be)

Maybe you can use the way you store the current angle. In my example above you see then on reopening the animation dialog the angle 74. This was obviously the last angle position and it is still available after closing the dialog. So maybe you can save the range values the same way.

Zolko-123, 2021-01-12:
that value is the **current** variable value

donovaly, 2021-01-12:
> that value is the **current** variable value

Ah, OK. However, you already store the state of the Loop and Pendulum property, so it should be possible to store the values of the range the same way, no?

Alonso-JAMM, 2021-01-12:
Maybe a dictionary could be used to store the value, range, step, and other data of a variable. For example, the Variables object could have a property "App::PropertyPythonObject" that would store a dictionary containing the animation data related to all variables. So the dictionary would look like this:
`{"angle1": {"value": 74, "Range_Begin": 0, "Range_End": 360, "step":1}, 
   "angle2":{"value": 10, "Range_Begin": 5, "Range_End": 20, "step":1}}`

Then the data stored in the dictionary could be used by the animation window. This would be helpful in storing the animation data of multiple variables at the same time

Zolko-123, 2021-03-25:
This should be fixed in the recent merge of Thowil's PR

commit fbcd8436 (2021-03-14): Use a document-property (simple dict) to save the configuration-values last used to animate each variable. (Should fix #161)
closed 2021-03-25 by Zolko-123