Zolko-123/FreeCAD_Assembly4 #162: variable description not visible or changeable

donovaly, 2021-01-11:
When creating a variable one can set a description. This is very useful when you have many variables. However, after the creation one cannot see the description. (maybe I have overseen something?)

It would be helpful that the description appears in the properties and maybe also as tooltip for the variable combobox in the animation dialog.

Zolko-123, 2021-01-12:
The description should be visible in the tooltip if you leave the mouse hover over the variable name.

As for adding a description later, it's a good idea, I have to check how to do it

donovaly, 2021-01-12:
> The description should be visible in the tooltip if you leave the mouse hover over the variable name.

This doesn't work for me:

[image]

I can also not see the description anywhere else. Maybe you could add a property holding the description?

Zolko-123, 2021-01-12:
did you try if you select "Variables" in the tree and hoover over the name in the property panel ?

donovaly, 2021-01-12:
> did you try if you select "Variables" in the tree and hoover over the name in the property panel ?

Not yet. But this works, thanks -> we need a proper documentation ;-)
As I wrote, I volunteer, but cannot yet estimate when I will find the time.

closed 2021-03-10 by Zolko-123