Zolko-123/FreeCAD_Assembly4 #163: [enhancement] variable name can be changed

donovaly, 2021-01-11:
When creating a variable, I made a typo. I realized it later and want to change the variable name but there is no way to do this.

So it would be a useful enhancement if one can change the variable name.

Zolko-123, 2021-01-12:
This has been asked before, but unfortunately it's not possible.

closed 2021-01-12 by Zolko-123

donovaly, 2021-01-12:
> This has been asked before, but unfortunately it's not possible.

What about this?: Use the way FC does this for e.g. Sketch and Part objects names. The internal name is fixed, but you can change the property Label. The Label can be referenced in formulas as `<<Label>>`.
So if I have for example the variable name "angel", I can later set its label to "angle". Then I can reference it as
`Variables.<<angle>>`.

Zolko-123, 2021-01-12:
yes, that would be cool, but it doesn't work like that: a property of a FreeCAD object doesn't have a label.

donovaly, 2021-01-12:
> yes, that would be cool, but it doesn't work like that: a property of a FreeCAD object doesn't have a label.

So on cannot add a property to a variable? Can you ask in the developer forum, maybe others like @wwmayer have an idea how this can be done or the situation be improved.

wwmayer, 2021-01-16:
> The internal name is fixed, but you can change the property Label. The Label can be referenced in formulas as <<Label>>.

_Label_ is a property and like for any other property you can change its value, but not its name.

When you want to change the name of a user-defined property then the only way is to remove it and re-add it using the new name:
```
doc = App.newDocument()
obj = doc.addObject("App::DocumentObject")
obj.addProperty("App::PropertyFloat","Angel")
obj.Angel
obj.removeProperty("Angel")
obj.addProperty("App::PropertyFloat","Angle")
obj.Angle
```

donovaly, 2021-01-16:
Many thanks for the info!

Zolko can you thus please reopen the issue? (It is of course not necessary to implement this soon, but very valuable to have the info that it could be done and how in an open enhancement.)

Zolko-123, 2021-01-16:
Werner confirmed that you cannot change the name of a user defined property, you have to delete it and create a new one. This is already possible today, why should this issue be re-opened ?

donovaly, 2021-01-16:
> Werner confirmed that you cannot change the name of a user defined property, you have to delete it and create a new one. 

Then I still did not get it. He wrote:
>> Label is a property and like for any other property you can change its value, but not its name.

So you can add a property _Label_ to the variable. Changing the variable name then changes the value of _Label_, no?

Zolko-123, 2021-01-16:
1) may-be, but I don't know how

2) this is no how I understand Werner's comment. May-be confirm with him, and ask for the corresponding code

wwmayer, 2021-01-19:
@donovaly You should provide an example of what exactly you do and want to achieve.

donovaly, 2021-01-21:
> @donovaly You should provide an example of what exactly you do and want to achieve.

Here is an almost empty FC file:
[Variable-renaming-task.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/5846508/Variable-renaming-task.zip)

The last entry in its model tree is "Variables". There you see a variable named "angle". I need now a way to fix my mistake and rename it to "angle".

I understood you that way that one could add the property "Label" to this variable. Then the user could change the Label property to effectively change the name of the variable (despite the internal name will be unchanged).