Zolko-123/FreeCAD_Assembly4 #164: Can't create lcs on the second and subsequent link  in the assembly model

AcumenDev, 2021-02-02:
Hello
I can't create lcs on the second and subsequent link  in the assembly model.

When I create lcs on Face not the first part it is created in the wrong place. 

Example project:
[check.zip](https://github.com/Zolko-123/FreeCAD_Assembly4/files/5914249/check.zip)

FreeCAD Assembly 4 workbench
Current version 0.9.13, 2020-12-01

OS: Ubuntu 20.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23756 (Git) AppImage
Build type: Release
Branch: master
Hash: 9c6e9184930a52b165a0b7274e3a45d1006bfe67
Python version: 3.8.6
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

Zolko-123, 2021-02-11:
Can you please describe more the exact steps that you're doing ?

AcumenDev, 2021-02-11:
> Can you please describe more the exact steps that you're doing ?
I recorded video 
https://youtu.be/fMjeIzxvw00

Zolko-123, 2021-02-11:
OK, I understand. Actually, you can't do that. Not even for the first one. This is not an Assembly4 issue but a FreeCAD issue: attachments don't work on link objects. At least not in v0.19

What you have to do is: 
- create the LCS in the part itself, not in the assembly. The best way for that is to hide the assembly and only show the part, to be sure to not select the wrong objects.
- then, in the assembly, select the LCS that you need, in the part instance that you want, and click "Import Datum". This will create an LCS in the assembly, and superimpose it to the original LCS in the part, in the correct instance (if you have several instances of the same part). 
- this LCS is now native in the assembly and can be used for construction and further assemblies, and it will always follow the original LCS if you edit the original part's parameters. 

This is the method you need to follow if you do multi-level assemblies

AcumenDev, 2021-02-11:
Thank you, it worked.

AcumenDev, 2021-02-12:
> OK, I understand. Actually, you can't do that. Not even for the first one. This is not an Assembly4 issue but a FreeCAD issue: attachments don't work on link objects. At least not in v0.19
>
> What you [...]

I have another question: how to create an lsc on two links at the same time?
I tried to import their lsc and then create a new lcs on the imported ones, but then the view breaks.

closed 2021-03-10 by Zolko-123