Zolko-123/FreeCAD_Assembly4 #165: Create new Fastener

Zolko-123, 2021-02-11:
When a fastener is selected, then the "Insert new fastener" should pre-set the parameters of the selected fastener for the new fastener. This way, chain-creating many fasteners would be much faster.

Alonso-JAMM, 2021-04-07:
Hello, I worked on this issue and I think now I have a working code for pre-setting parameters of new fasteners. There are two ways of setting the parameters of new fastener. The first one is by selecting the fastener to copy. For example, if you select a screw and the press the "Insert screw" button, then a new screw with the same parameters will be created. The second way is by just clicking the button without selecting; this will create a new fastener with the same parameters of the old fastener if there is a previous fastener of the same type than the type of the new fastener. For example, if you create a screw and set the parameters of this screw and then you click the "Insert screw" button, then the parameters of the previous screw will be copied to the new screw.

Zolko-123, 2021-04-08:
thank-you, I had computer problems for some time and couldn't look at your proposal. I'll do it ASAP in the coming days.

Zolko-123, 2021-10-04:
is this still relevant ?

closed 2021-10-04 by Zolko-123

xoviat, 2021-10-04:
I would like to see this. I can take @Alonso-JAMM's code forward if you want.

Zolko-123, 2021-10-04:
@xoviat : that would be nice, but don't you want to finish the imperial unit spinbox first ?