Zolko-123/FreeCAD_Assembly4 #166: symetrical part

Zolko-123, 2021-02-11:
To create a symmetrical part of an inserted part (link):

- Create a new App::Part, ask user about it's name and symmetry plane (XY, XZ, YZ)
- place that part into the Parts folder
- Create a Draft_Clone of the chosen App::Link (inserted part)
- Move that clone into the App::Part
- Create a Part::Mirroring (need to ask about which plane to make symmetry. Probably in the first dialog)
- Move that mirror to the App::Part
- hide the clone
- Launch Asm4 "Insert Part" with that symmetrical part

rolfbachmann, 2021-02-25:
Hi, I need a simple mirrored body in my project and I found this issue which really helps me.

What I don't understand is the clone?

1. I've a body (BodyLeft)
2. In in the "Part" workbench I create a mirror which gives me a "BodyLeft (Mirror #1)"
3. Then I create a new Part (BodyRight), put it in App::Part
4. I put the "BodyLeft (Mirror #1)" into the Part (BodyRight)
5. Now I can launch Asm4 and do "Insert Part"

Are there any issues/side effects when I do it like this, without the clone?

Zolko-123, 2021-02-25:
Hi, good catch, I don't know. May-be the intermediate step of creating a clone is unnecessary. It would be nice if you could report after your test.

rolfbachmann, 2021-02-25:
Hi, again. Nop. I did a short test and it didn't work. The clone is needed. Otherwise the base part of the mirror (BodyLeft in my case) is not available on the "Insert Part" dialog anymore.

Zolko-123, 2021-02-25:
OK, thanks for reporting

Zolko-123, 2021-03-10:
This is now done in v0.9.15. A Link is used instead of a Draft.clone.

closed 2021-03-10 by Zolko-123