Zolko-123/FreeCAD_Assembly4 #168: Import Datum

Zolko-123, 2021-02-11:
the command "Import Datum" should import that datum in the higher level App::Part (or GeoFeatureGroup" and not only in an Asm4 Model. Thus, this could work on imported STEP files if they have many nested parts

diniciacci, 2021-04-13:
Could this open up basic quasi-sub-assembly in one document workflow?

If Import Datum would work on a LCS located inside of a `Body` that is inside of a `Part`:
[image]
It would then be able to place the `Part` (Carter Assembly) in the assembly relative to some LCS that is in the `Body` (Carter). Currently this is not possible because LCS in `Body` cannot be used for assembly or imported. The alternative of creating LCS in `Part` (Carter Assembly) cannot be used because LCS cannot reference a contained `Body` (Carter) else they fall out of scope.

Importing the `Body` LCS would allow this workflow (and the converse, placing other `Part`s in the assembly relative to nested LCS after importing them, right?)

An alternative that would allow the same workflow would be to allow importing a Link to a LCS contained in a `Body` inside a `Part`:
[image]
Currently the "Import Datum" is not possible for links to my knowledge. That would again allow to link to LCS that are contained inside a `Body` nested inside a `Part` hence allowing a quasi-nested assembly. Links overcome the "fall out of scope" and allow for clean `Body` <-> `Part` crossing and import datum on links allows for using them in Assembly 4 effectively unlocking some basic nested positioning.

This is on Version 0.9.15.

Zolko-123, 2021-08-23:
this has been improved in latest v0.10, please test and report

closed 2021-10-04 by Zolko-123