Zolko-123/FreeCAD_Assembly4 #186: Crash on import

markstory, 2021-03-24:
Similar to #73 I'm also having a crash when attempt to link a part. I could be doing something very wrong though as I'm a FreeCAD noob.

I'm running FreeCAD 0.18 (from flatpak) and installed Assembly4 via the addon manager this evening.

<details>
<summary>Expand for full logs</summary>
<pre>
ᐉ flatpak run org.freecadweb.FreeCAD
FreeCAD 0.18, Libs: 0.18R14555 (Git shallow)
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2019
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Assembly4 workbench loaded 
Gtk-Message: 00:06:11.848: Failed to load module "appmenu-gtk-module"
Gtk-Message: 00:06:11.873: Failed to load module "canberra-gtk-module"
Gtk-Message: 00:06:11.873: Failed to load module "canberra-gtk-module"
jDocument with the UUID '97bc75a8-3d1f-4ca4-b3fc-8b565fa3109a' already exists, change to '4b4e5a7d-e27e-4f05-aae9-d0e061c2cce3'
Assembly4 WorkBench initializing ......................... done.
Activating Assembly4 WorkBench
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x39880) [0x7f55b25f5880]
#1  0x7f55b3dd9798 in App::OriginGroupExtension::relinkToOrigin(App::DocumentObject*) from /app/freecad/lib/libFreeCADApp.so+0x38
#2  0x7f55b3dda35f in App::OriginGroupExtension::addObjects(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> >) from /app/freecad/lib/libFreeCADApp.so+0x4f
#3  0x7f55b3d8cf7a in App::GroupExtension::addObject(App::DocumentObject*) from /app/freecad/lib/libFreeCADApp.so+0x8a
#4  0x7f55b3d8cb50 in App::GroupExtension::addObject(char const*, char const*) from /app/freecad/lib/libFreeCADApp.so+0x70
#5  0x7f55b3d8fd75 in App::GroupExtensionPy::newObject(_object*) from /app/freecad/lib/libFreeCADApp.so+0x65
#6  0x7f55b3d9034e in App::GroupExtensionPy::staticCallback_newObject(_object*, _object*) from /app/freecad/lib/libFreeCADApp.so+0x3e
#7  /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0xb9597) [0x7f55b3728597]
#8  /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyObject_MakeTpCall+0xa7) [0x7f55b3754937]
#9  /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x81f7) [0x7f55b36e7d27]
#10  /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x6ff23) [0x7f55b36def23]
#11  /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x16c2cb) [0x7f55b37db2cb]
#12  /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0(PyVectorcall_Call+0x6f) [0x7f55b372905f]
#13  0x7f55b17683c1 in PySide::SignalManager::callPythonMetaMethod(QMetaMethod const&, void**, _object*, bool) from /app/lib/libpyside2.cpython-38-x86_64-linux-gnu.so.5.15+0xa1
#14  0x7f55b17687b0 in PySide::SignalManager::qt_metacall(QObject*, QMetaObject::Call, int, void**) from /app/lib/libpyside2.cpython-38-x86_64-linux-gnu.so.5.15+0x110
#15  /usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2ff100) [0x7f55b2c9b100]
#16  0x7f55b3234dd2 in QAbstractButton::clicked(bool) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x42
#17  /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x26873a) [0x7f55b323573a]
#18  /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x26a08f) [0x7f55b323708f]
#19  0x7f55b3237261 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xf1
#20  /app/lib/python3.8/site-packages/PySide2/QtWidgets.cpython-38-x86_64-linux-gnu.so(+0x585357) [0x7f5593aed357]
#21  0x7f55b3187f9e in QWidget::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1ce
#22  0x7f55b314815f in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x7f
#23  0x7f55b314f04b in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x49b
#24  0x7f55b43bdbe0 in Gui::GUIApplication::notify(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0x90
#25  0x7f55b2c64dda in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12a
#26  0x7f55b314e076 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1d6
#27  /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1d2fe1) [0x7f55b319ffe1]
#28  /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1d630e) [0x7f55b31a330e]
#29  0x7f55b314815f in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x7f
#30  0x7f55b43bdbe0 in Gui::GUIApplication::notify(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0x90
#31  0x7f55b2c64dda in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12a
#32  0x7f55b0d8a443 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0x633
#33  0x7f55b0d6056c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#34  /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x6548a) [0x7f55acb1948a]
#35  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x25b) [0x7f55ae92285b]
#36  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x55b08) [0x7f55ae922b08]
#37  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7f55ae922bd3]
#38  0x7f55b2cbc32f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x5f
#39  0x7f55b2c6379b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#40  0x7f55b2c6ba10 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x90
#41  0x7f55b434c216 in Gui::Application::runApplication() from /app/freecad/lib/libFreeCADGui.so+0x1a36
#42  /app/freecad/bin/FreeCAD(main+0x6c0) [0x5642eb98bd30]
#43  /usr/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf2) [0x7f55b25e0062]
#44  /app/freecad/bin/FreeCAD(_start+0x2e) [0x5642eb98c08e]
</pre>
</details>

Zolko-123, 2021-03-25:
> I'm running FreeCAD 0.18 (from flatpak) and installed Assembly4 via the addon manager this evening.

Assembly4 is not compatible with FreeCAD 0.18, needs at least v0.19

closed 2021-03-25 by Zolko-123

markstory, 2021-03-25:
Thanks for the quick reply, I'll try with a newer FreeCAD release.