Zolko-123/FreeCAD_Assembly4 #188: [enhancement?] Showing only "usable" LCS

rponthieu, 2021-04-01:
Hi,

I was wondering if it could be interesting to only show the LCS that can be used for attachment in the 3D view ? (the ones at the root of parts imported).

By default all the LCS are shown, it is less easy to work with assemblies of assembles. 
I guess it can also participate to the confusion of beginner when some are shown but cannot be used (as in issues #98 #84 #72 #121 where the solution was to use import datum).

I don't know however if it is an easy thing to modify.

Best,
R

Zolko-123, 2021-04-08:
by "showing" you mean in the 3D window or in the drop-down list ? Hum, anyway, this is a good idea, let me think about it.

rponthieu, 2021-04-08:
I meant in the 3D window yes.

Zolko-123, 2021-06-25:
This would force to hide/unhide existing LCS, remembering and then restoring their visibility state: too error-prone for a small usability comfort. Would be nice to have, but too risky.

closed 2021-06-25 by Zolko-123