Zolko-123/FreeCAD_Assembly4 #189: Select an Attach to Parent inside a Group

diniciacci, 2021-04-13:
#### Use case
I'm working on a use case where I have all the Parts in an unique file and I'm assembling them with Assembly 4:

[image]

#### Problem
In this use case the main issue is keeping tidy, since you can't have sub-assemblies (without breaking the use case of having one file) it's easy to end up with a long list of Parts (for example all the bolts in the assembly). It would be useful visually to be able to group items (e.g. all the bolts of a certain type). Currently this is not (completely) possible. The problem is that when you create a `Group` in the Model and move items inside it then you cannot select them anymore as Parents to Attach To.

[image]

The PARENT_PART001 is not present in the select "Attach to:" because PARENT_PART001 is inside "Group":
[image]
[image]

#### Proposed solution
Just present in the select all the contents of Group s inside of Model in a flat fashion.

#### Note
After you move them inside a `Group` moving them outside is rather hackish but this is likely unrelated to Assembly 4 and the "Group reordering trick" on the forum solves it and you can take them back out.

This is relevant because after you finish the model you can move the Parts in the Model inside of a group (the Parent relationship will naturally break, but they will stay at their position, not auto updating).

#### Relevant error
If "Insert a link to part" is pressed when a `Body` inside a Group is selected the following error is printed:
Running the Python command 'Asm4_insertLink' failed:
Traceback (most recent call last):
  File "/home/gab/.FreeCAD/Mod/Assembly4/insertLinkCmd.py", line 77, in Activated
    if parent.TypeId == 'App::Part' and parent.getParentGeoFeatureGroup() is None:

'NoneType' object has no attribute 'TypeId'Running the Python command 'Asm4_insertLink' failed:
Traceback (most recent call last):
  File "/home/gab/.FreeCAD/Mod/Assembly4/insertLinkCmd.py", line 77, in Activated
    if parent.TypeId == 'App::Part' and parent.getParentGeoFeatureGroup() is None:

'NoneType' object has no attribute 'TypeId'

Zolko-123, 2021-06-25:
This has been requested several times, but somehow I feel that using groups in assemblies is a poor-man's sub-assembly: why not create a sub-assembly that would contain the parts that you regroup into a group ? If we do what you suggest, what about groups in a group ? Where does it end ? 

My preferred solution to this problem would be to be able to create sub-assemblies from selected parts in an assembly: but that is quite complex.

diniciacci, 2021-06-25:
The limit for our use case is that we want to have only one file, sub-assemblies require different files if I remember correctly.

In our workflow (university and/or student) managing multiple files is slightly prone to errors (i.e. might look silly, but immagine 20-100 students having to deliver two or three files instead of one, looks like a minor thing but the time invested in organising them become relevant fast).

Another minor point is that sometimes we do not use groups for sub-assemblies, but for example we want to group fasteners of same type or components that visually we want to toggle (make them appear or disappear) all together (i.e. locate them fast for clicking). In this second case those "carter" parts do not always make sense as an assembly. I.e. maybe they are not even attached together, but they are indeed related conceptually because they are all "covers".

Imho just ignoring groups and listing their recursive contents as a flat list could be enough. The part browser gets tidy and the item can still be selected like group does not exists.

Your solution is clearly the way to go, but consider that it would not clash with this modification imho.

s-em-i, 2022-02-01:
@diniciacci,
I agree with you. As I ran into the same Issues, linked Parts within Groups are not getting listed in the **Place linked Part**
GUI. But they get listed in the **Insert a Part** GUI. I tried the sub-assemblies work around as well, but it didn't work
for me either. Contradicting to @Zolko-123 opinion, "using groups in assemblies is a poor-man's sub-assembly",
I see it the way as FreeCAD sees it:
[image]

Nevertheless, I do agree with @Zolko-123 about groups in a group and where does it end ?
So I made some changes in @Zolko-123 awesome Workbench, and please consider this as a Pull Request as well.

I changed **[placeLinkUI.py](https://github.com/Zolko-123/FreeCAD_Assembly4/blob/b57a811eefc21c9d751746703af41f1356d70087/placeLinkUI.py#L134-L139)** @ line **134**
```
	# find all the linked parts in the assembly
	for objName in self.rootAssembly.getSubObjects():
		# remove the trailing .
		obj = self.activeDoc.getObject(objName[0:-1])
		if obj.TypeId=='App::Link' and hasattr(obj.LinkedObject,'isDerivedFrom'):
```
to
```
# find all the linked parts in the assembly
for obj in self.activeDoc.findObjects("App::Link"):
	if self.rootAssembly.getObject(obj.Name) is not None:
		if hasattr(obj.LinkedObject,'isDerivedFrom'):
```
Now it finds every App::Link inside rootAssembly, even if it is in a group, or sub-group, or sub-sub-group ... No Limit is set.
The for loop finds EVERY App::Link in the working Document. And If this App::Link can be found in the main working
Assembly, it is a possible candidate. I guess the actually Issue lies within the getSubObjects() function. I would
expect it return all the Sub-Objects, but apparently only the ones from the first level. And no search depth can be passed.

And in order to get the Place_Link Icon active, If you select an App::Link within a group,
I changed **[placeLinkCmd.py](https://github.com/Zolko-123/FreeCAD_Assembly4/blob/b57a811eefc21c9d751746703af41f1356d70087/placeLinkCmd.py#L42-L43)** @ line **42**
```
# something at 2nd level is selected
if tree and len(tree)==2:
```
to
```
# something at 2nd+ level is selected
if tree and len(tree)>=2:
```

I have to admit, I've only checked a few features with this mod, but all the ones I need for my project are
working with this.
Cheers
Semi

.

diniciacci, 2022-02-02:
@s-em-i Personally I like the solution you implemented. Rereading my post I should have highlighted more that by "flat fashion" I intended what you nicely implemented. Very good and well done IMHO. I yet have to try this change, but it looks like it does fix my use case.

Zolko-123, 2022-02-03:
> So I made some changes in @Zolko-123 awesome Workbench, and please consider this as a Pull Request as well.

Thnax. I have tried some things but your code is better. 

> Nevertheless, I do agree with @Zolko-123 about groups in a group and where does it end ?
> ```
> # something at 2nd+ level is selected
> if tree and len(tree)>=2:
> ```
what about:

```
elif len(tree)==3:
    if App.ActiveDocument.getObject(tree[1]).TypeId=='App::DocumentObjectGroup':
```

This should only allow 1 level of groups. Now the same needs to be done for fasteners

Zolko-123, 2022-02-03:
pushed an update in the development version. Please test and report

s-em-i, 2022-02-03:
@Zolko-123,
thank you for taking this in so quickly, very much appreciated.
The changes in the placeLinkUI.py are working as expected.

The changes in the placeLinkCmd.py don't. Well they do work almost as you described it.
Lets go through this step by step,
here are some screenshots of your mod:
**rootAssembly** selected -> Place_Link Icon = **inactive** -> right!
[image]
**Group** selected -> Place_Link Icon = **inactive** -> right!
[image]
**PartInGroup** selected -> Place_Link Icon = **active** -> right!
[image]
**SubGroup** selected -> Place_Link Icon = **active** -> wrong!
[image]
**PartInSubGroup** selected -> Place_Link Icon = **inactive** -> right!
[image]
**SubSubGroup** selected -> Place_Link Icon = **inactive** -> right!
[image]
**PartInSubSubGroup** selected -> Place_Link Icon = **inactive** -> right!
[image]

Conclusion:
This mod doesn't work as intended, and it restricts the users freedom of development style.

---
here are some screenshots of my mod:
**rootAssembly** selected -> Place_Link Icon = **inactive** -> right!
[image]
**Group** selected -> Place_Link Icon = **inactive** -> right!
[image]
**PartInGroup** selected -> Place_Link Icon = **active** -> right!
[image]
**SubGroup** selected -> Place_Link Icon = **inactive** -> right!
[image]
** PartInSubGroup** selected -> Place_Link Icon = **active** -> right!
[image]
** SubSubGroup** selected -> Place_Link Icon = **inactive** -> right!
[image]
** PartInSubSubGroup** selected -> Place_Link Icon = **active** -> right!
[image]

Conclusion:
This mod does work as intended, and it doesn't restricts the users freedom of development style.

To be honest, I don't see the sense in restricting your code that much, especially by this only allow 1 level of groups
policy. My Version with just the ">=" mod, works very well. The restrictions for the App::Part are still met,
and it doesn't inflate the code. My advise in a general speaking matter, keep it simple but versatile.
Sure, one could say not many people are using this sub-sub-sub-group thing, I don't use it. But why
restricting anybody's development style? There is also much more code to maintain if you are putting
too much restrictive code in it.

While browsing through your code, I am trying to fix some show/hide LCS issues, I figured the actually
issues are coming from too much restrictions and the lack of finding ALL the objects you need to investigate. 

I don't get the idea with the fasteners? What needs to be done there as well?
When I try to add a Fastener, they behave like every other part in your workbench.
I usually don't use them, I am using imported step files for my hardware.
If you are having some kind of Test Bench Check List to go through, I am happy to use it.

btw:
Can you tell me please the correct syntax, how to add your Workbench with the development branch
into the custom repositories?
[image]

```
https://github.com/Zolko-123/FreeCAD_Assembly4
development
https://github.com/Zolko-123/FreeCAD_Assembly4.git
development
```

Zolko-123, 2022-02-03:
updated with your version. seems to work. 

And I don't know how to add the development branch as default Addon repository, sorry (I use my local version, obviously)

s-em-i, 2022-02-03:
Thank you too!
It seems the branch feature is still in development state.
I had to change some code to get it working.
If I check the source code on the 20th Version of Freecad, it is already fixed on those points.
I guess this issue can be closed now.

Zolko-123, 2022-02-03:
> I don't get the idea with the fasteners? What needs to be done there as well?

did you try to attach a fastener to a part (link) in a group ?

s-em-i, 2022-02-03:
> > I don't get the idea with the fasteners? What needs to be done there as well?
> 
> did you try to attach a fastener to a part (link) in a group ? Somehow it doesn't work and I don't know why.
Yes, seems to work like usual!

Zolko-123, 2022-02-03:
merged in master

closed 2022-02-03 by Zolko-123