Zolko-123/FreeCAD_Assembly4 #197: Automatic LCS creation + 3D Visualisation

jdorel, 2021-05-05:
Right now, a big part building an assembly is adding the reference LCS to the parts. Would it be interesting to automatically add them to the parts ?
One issue with that would be the match between LCS, because as they would be created automatically, the naming might not easily be recognisable.
Here is one way I think matching LCS could be done : a left pane with the linked part showing it's LCS and another left pane with the receiving model LCS. This is essentially the current text pane, but in 3D. It would make the experience a lot more intuitive (and fast).

Zolko-123, 2021-05-06:
yes, this would be nice, and this is how T-Flex CAD does it, but it's way beyond my coding capabilities

Zolko-123, 2021-08-23:
This is for now out of the possibilities.

closed 2021-08-23 by Zolko-123