Zolko-123/FreeCAD_Assembly4 #198: Question: New Hole LCS Static Name

GeekinaCave, 2021-05-21:
Good Day

I noticed when you create a new datum, all options show a text box with a predeterminated name that you can change however you want before create said datum; except "New Hole LCS" that directly puts HoleAxis_# in the name. Is this way by design?.

Zolko-123, 2021-05-24:
It's by laziness. Do you think it would make sense to give the user a chance to change the default name ?

GeekinaCave, 2021-05-24:
Well, to my OCD definitely would. 😅 
Mostly because if you change the name later; it shows both the new (and old one) in the assembly menu.

Zolko-123, 2021-10-04:
Since v0.9.14, many HoleAxis datums can be created in 1 step, which is a great improvement I think, and then it's not possible to make user-adjustable names, sorry.

closed 2021-10-04 by Zolko-123