Zolko-123/FreeCAD_Assembly4 #208: Error "No module named 'AnimationLib'"

cyclops1982, 2021-07-04:
Hi,

i"ve just downloaded the latest Freecad from the main website (FreeCAD_0.19-24291-Linux-Conda_glibc2.12-x86_64.AppImage). Started it, and then installed Assembly4. The install went fine.

I have tried to switch to this workspace but get the following error:
```
cyclops@w540-cyclops:~/Downloads$ ./FreeCAD_0.19-24291-Linux-Conda_glibc2.12-x86_64.AppImage 
FreeCAD 0.19, Libs: 0.19R24291 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2021
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Assembly4 workbench loaded 
No module named 'AnimationLib'
Traceback (most recent call last):
  File "<string>", line 111, in Initialize
```

Zolko-123, 2021-07-04:
yep, sorry, was a stupid mistake (due to fucking GIT, do I hate that crap).

Can you please update the workbench ?

Zolko-123, 2021-07-04:
I tried and I think it's fixed.

cyclops1982, 2021-07-05:
GREAT!

Thank you. It works now!

closed 2021-07-05 by cyclops1982

cyclops1982, 2021-07-06:
Hey @Zolko-123  

Just want to send you an additional message. I'm fairly new to FreeCAD so still struggling a bit, but this seems to be a lot better than what i've seen before a few years ago with regards to assembling! Thanks for the work!