Zolko-123/FreeCAD_Assembly4 #211: Enhancement: Relink moved files

dannystaple, 2021-07-24:
I reorganised some files in FreeCAD projects into a FreeCAD parts library. When doing so, this broke links in some assembly4 models.

I have written up how I was able to solve this in the python console - https://orionrobots.co.uk/robot%20building/2021/07/24/freecad-assembly-link-issue.html. 

Could I suggest a GUI option, with the following workflow, to remedy this:
* Start with the assembly, and the parts loaded.
* An option either in the context menu or the broken link, or on the menu bar starts a relink dialogue.
* Pick the part to relink. Assuming the reference geometry is the same - it should work.

What I've not considered is what happens if a user selects a part with different geometry and datums, how to behave or prevent that.

Zolko-123, 2021-07-25:
yes, good catch.

Actually, there is a built-in function that allows to re-link lost files: in the "LinkedObject" property field, click on the small button with ... on it. This will open a dialog where you can pick another object to link-to.

BUT: this is not very conveient, as this allows to link to any object, even those that Asm4 doesn't allow to link to.

SO: what would you think about the following procedure: if a link is selected in the assembly, click on the "Insert Link" button, and when this detects that the link is broken (obj.LinkedObject == None) then it will open the link UI and allow to select a - Asm4-valid - object (Part or Body).

Currently, when selecting a linked object, Asm4 proposes to duplicate the link, which I find handy ... but which fails if the link is broken. We could think of adding an extra check about the LinkedObject property.

What do you think ?

luzpaz, 2021-07-25:
@Zolko-123 ping

Zolko-123, 2021-07-26:
is the the development branch

luzpaz, 2021-07-26:
@Zolko-123 Thanks for responding. Sorry my ping came in after you responded :blush:  
Can you label this ticket "waiting user response" so we can track this? thanks!

Zolko-123, 2021-08-23:
done in latest v0.10, please test

closed 2021-08-23 by Zolko-123