Zolko-123/FreeCAD_Assembly4 #212: Option to rename Variables

BloodyRain2k, 2021-07-28:
I really like the easy adding of variables that Assembly4 offers, but it seems that there's no way to rename a variable?

I mistyped a variable's name during creation and I couldn't find a way to rename it, and I also didn't want to delete it to because that would likely have broken the points where I used it.

So could the dialog for deleting a variable get a button added that offers renaming it instead?

As in, you select the variable you want to rename the same way you would select it for deleting it, but instead of then clicking the delete button, you change the name and click a new button like "Update" or something like that.

Zolko-123, 2021-07-30:
This is a core FreeCAD limitation, so not possible

closed 2021-07-30 by Zolko-123

sberbinschi, 2024-04-26:
It's an old post, but I can give you a hint. I had the same problem and I just create a new variable with the correct name, then I assigned it all relations where the old one is used, and then I deleted the incorrect one. That's all.