Zolko-123/FreeCAD_Assembly4 #213: It looks like this assembly is not working fine anymore.

leoheck, 2021-08-22:
Hi, it looks like this assembly is not working fine anymore. 

The `Placed Linked Part` does not show `Selected LCS in Part` most of the time.
`Ignore` and `Cancel` on top of this panel also do not work fine. Both add the Model anyways.
Ok button sometimes shows `16:15:52  Problem in selections`
Even that, the `Ignore` and `Cancel` place the part in the assembly.

I am adding the LCS inside the Body. Some bodies were not created with the Body tool on Assembly4 toolbar.
Do these Body's are different? If they are, is it possible to have a mark or a different color on them so the user does not get confused?

Also, should I but Body inside a Part and then create th LCS inside the Part?

These are my objects, for instance.
[image]

I am using this version.
```
OS: Ubuntu 21.04 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.
Build type: Release
Branch: unknown
Hash: 378dec720c808f5b05216ed5e890e6d49bdc999a
Python version: 3.9.5
Qt version: 5.15.2
Coin version: 4.0.0
OCC version: 7.5.2
Locale: English/United States (en_US)
```

Zolko-123, 2021-08-23:
yes, something strange happened with FreeCAD that broke the ExpressionEngine. I'm trying to figure out.

> Some bodies were not created with the Body tool on Assembly4 toolbar. Do these Body's are different?

They're the same.

leoheck, 2021-08-23:
Nice. Thank you for the feedback @Zolko-123 is there anything I can do to assist you?

Zolko-123, 2021-08-23:
actually, yes: I just pushed a new version, a big upgrade (the first in the v0.10) and if you could update your workbench and test whether this corrects the behaviour. 

And also please report any stuff that doesn't work.

leoheck, 2021-08-23:
Sure! It will be my pleasure. I just have to say I am not fully aware of how to use the workbench yet in the right way. But I will test it for sure. Thank you.

leoheck, 2021-08-23:
These are the buttons now.

For me, Apply means Ok.
Cancel is not canceling anything (it is still placing the link/object)
Ok - Shows this message `11:31:42  Problem in selections`

[image]

Let me guess what you were trying to do.

Apply - Same as ok?

Cancel - should cancel this operation, it cannot add anything to the object tree. The Place Linked Part action should be canceled.

Ok - should be the Place linked Parts was properly configure so everything needs to be well configure to end/close this Place linked Part action.

leoheck, 2021-08-23:
It would be nice if I could double click an object here in this list. 

[image]

=======

When I try to insert the model for the first time, it does not see the LCS of my part.

[image]

=======

After some tries, it starts to show its LCS but the OK button still tells me there is an issue. Maybe I have to select LCS even if I have a single one on each list.

[image]

======

Actually, sometimes the Apply doesn't do anything. In this test I dont have any LCS listed in both lists.

leoheck, 2021-08-23:
It looks like it is not working still. I cannot make it show the LCS of the selected object anymore.
I feel that the cancel button should cancel and not put the object in the Object tree too.

Zolko-123, 2021-08-23:
> For me, Apply means Ok.
> Cancel is not canceling anything (it is still placing the link/object)

Apply means OK+Close: this way, you can experiment with different attachment LCS. It's also useful when you need to change the orientation of offsets of a part: you select the attachment LCS (in both columns), hit Apply, and then adjust orientation and offsets. When you're happy, you click OK. 

Cancel cancels the **placement** of the part, not the **insertion**. If you want to cancel the insertion, you have to hit the "Cancel" button in the "Insert Part" dialog. When you place a part for the first time, Cancel only closes the dialog. If you re-do the placement of a part, Cancel cancels the re-placement and re-attaches the part where it was before.

Zolko-123, 2021-08-23:
> When I try to insert the model for the first time, it does not see the LCS of my part.

hum ... this is a **real** problem. Can you please describe the exact steps that you're doing to get this ? Are you sure there are LCS at the root of the Part/Body ?

leoheck, 2021-08-23:
> OK means Apply+Close: this way, you can experiment with different attachment LCS. It's a

Good. I think you said it wrong. But I feel you wanted to say:
```
Apply is to apply the current settings and remain in the config window.
And Ok is to Apply and Close the window.
```

Am I right?